bareboat-necessities/lysmarine_gen #13: Two finger touch zoom doesn't work with OpenCPN

mgrouch, 2021-02-23:
While I was able to implement a fix it required use of unreleased upstream libraries.

WxWidgets need to release version 3.2
and OpenCPN has to be compiled with it and with GTK3 support

More on the fix and existing patch from me

https://github.com/OpenCPN/OpenCPN/issues/2057

mgrouch, 2021-02-23:
Not a release stopper

tbultel, 2021-08-10:
Hi, using your master branch, I have also noticed the weird pan event, received instead of the zoom event,
when using 2 fingers. The zoom event only comes when using 3 fingers. Did you find out why ?

mgrouch, 2021-08-10:
I remember that was on gtk2 and not on gtk3

tbultel, 2021-08-11:
Hi, I confirm this happens with gtk3 as well. Reading the code of wxwidgets tells us that they apparently do the things the right way, and, moreover, their "event" test sample for gesture works like a charm. They do no use opengl, though.

tbultel, 2021-08-11:
By the way, despite of the need for 3 fingers, I have managed to get a better user experience of zooming, with that implementation of the OnZoom callback. The idea is that the given facture from gesture event is always relative to gesture start position, and has to be converted to something relative to the current level of map zoom:

void ChartCanvas::OnZoom(wxZoomGestureEvent& event)
{

    static double vps=1.0;
    double factor = event.GetZoomFactor();
    wxPoint pos = event.GetPosition();

    if (event.IsGestureStart()) {
        vps = GetVPScale();
    }
   
    double current_vps = GetVPScale();

    double wanted_factor = vps/current_vps*factor;

    ZoomCanvas( wanted_factor, true, false );

}

The static variable should be moved to a data member, of course. Next improvement is to have the zoom centered on the middle of the 2 fingers.

I also wanted to say that this zooming feature is a key one, as far as I am concerned. Even when setting the "+" and "-' buttons biggers, I do no find them convenient, when sailing, and most people are accustomed to zoom with fingers nowadays, due to smartphones (and that is what the android opencpn version does) .

tbultel, 2021-08-11:
Update: modifying the "penguin" sample in wxwidgets, to add gesture events, works fine (pan with 1 finger, zoom with 2)

tbultel, 2021-08-12:
FYI, I finally managed to have everything working, including the context menu without the need for an external program. 

https://github.com/OpenCPN/OpenCPN/pull/2333

mgrouch, 2021-08-31:
I got it working even with un modified 
OpenCPN with latest twofing. Needs more testing on BBN OS.

mgrouch, 2021-09-02:
Fixed in BBN OS 2021-09-01

closed 2021-09-02 by mgrouch