bareboat-necessities/lysmarine_gen #71: BerryGPS-IMU V4 - Understanding how to make it work with BBN OS

Tiki26, 2021-05-02:
I seem to have continual problems trying to get the Berrygps-IMU V4 working with either OpenPlotter or BBN OS. I warn you though, I'm brand new to the pi and need step by step instructions if there is a remedy. Sorry. Under OpenPlotter using I2c and address 0x77' baro was working ok. GPS via serial was also easy to setup. 
 But I can't find where to add or select sensors suitable for the Berry. As for Pypilot, I would love the digital compass and will consider an autopilot in the future but my  sensors for Pypilot didn't work with OpenPlotter or BBN OS.
I LOVE BBN OS. It's exactly what I have been looking for. I just love among other things, the abi!ity to dim the screen then to red. Just brilliant.
Hope you can help. Thanks
Regards

mgrouch, 2021-05-03:
I have  Berrygps-IMU V4 working in my setup. 

1. Install the hat
2. Follow steps to enable i2c and disable serial port https://ozzmaker.com/berrygps-setup-guide-raspberry-pi/
3. Reboot
4. sudo i2cdetect 1 (should show you addresses)
5. Create pypilot connection to signalK  (see  https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_imu  )
6. Set up barometer feed from SignalK ( https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_barometer_temperature_humidity )
7. IMU data heading etc should come from pypilot NMEA 
8. check it with telnet localhost 20220

mgrouch, 2021-05-03:
Brief explanation

1. i2c should be enabled, serial console disabled in config.txt
2. i2c driver should be loaded at boot (that's what raspi-config step does)
3. At this point you should have data readable from GPS (via /dev/serial0) and IMU / barometer (via  i2c)
4. Now you set up routing of this data into dashboards, chartplotters
5. IMU is read by pypilot, which feeds it via 20220 tcp port using NMEA 0183 format
6. pypilot also needs to have connection via SignalK web socket to port 80 of signalK
7. That connection needs to be authorized made READ/WRITE in signalK using their token exchange procedure
8. GPS is read by signalK by creating NMEA connection to /dev/serial0 port (set correct baud rate)
9. Barometric/temp data is read by SignalK using signalK BME280 plugin. Make sure setting polling interval below 30 seconds (because OpenCPN expires non-navigational data every 30 seconds)

Calibrate level IMU on water. Calibrate your IMU compass, using pyPilot calibration utility.

mgrouch, 2021-05-03:
Also this FAQ for common berryGPS issues

https://ozzmaker.com/berrygps_imu-faq/

Tiki26, 2021-05-04:
Thank you Mikhail for responding so quickly and thoroughly.
I had tried most of that but there are some new ideas for me. I had not known about the websocket requirement so I'll have to figure that bit out.
Also worth noting, the V4 Berry has a BMP388 sensors for pressure and temp. It worked in OpenPlotter I2C interface by using the BM3XXX sensor/driver. It will not work with the BME280. Is there a way to get the BM3XXX to work with your OS?

I'll report back as soon as I'm able to work through your list. Especially all the telnet checking methods...another thing I didn't know.
Cheers

mgrouch, 2021-05-04:
You need to check with SignalK folks for SignalK support of BMP388. 
I do not have that hardware. Probably easy to add support into SignalK based of Arduino python libraries but there is no way for me to test.

Tiki26, 2021-05-04:
Ok thank you Mikhail. I'm following a lead at the moment. If it works out
I'll share the details with you

mgrouch, 2021-05-04:
https://learn.adafruit.com/adafruit-bmp388-bmp390-bmp3xx/arduino

Tiki26, 2021-05-05:
I think getting the Version 4 Berry to work will require skills I don't have.
I will contact ozzmaker and ask if they have a version 3 unit still available as that one is clearly working for you.
Thank you for your very efficient and rapid support for what I think, is a great step forward in a boating OS.
Please close off the ticket.
Cheers

mgrouch, 2021-05-05:
Another approach is to ask openplotter to add support for BBN OS. Like I tried to do in

https://github.com/openplotter/openplotter-settings/issues/14

I do not think though openplotter developers even installed and evaluated BBN OS though. 
Or considered as another platform for openplotter scripts. It does make many of openplotter scripts obsolete but some could be still useful.

Tiki26, 2021-06-28:
Hi Mikhail

Just wanted to let you know I managed to get Mark from Ozzmaker and Sean
together to work out a solution to getting the Berrygps-IMU V4 to work with
Pypilot. It took them a couple of weeks but they've done it. Full support
will be built in to the next OpenPlotter image but in the meantime Mark
will put the workaround on his forum.
This now at least, gives people an option to buy a great commercial product
relative!y cheaply that rivals the Moitissier Hat (minus AIS).
I'll send you a link to Marks sticky post when I'm made aware of it.

Thanks

Regards

Neil Rogers
PS while I have a small amount of Sean's attention, I'll lobby for him to
look at BBN OS

mgrouch, 2021-06-29:
Thanks a lot for helping on this.
PyPilot fixes will take care of support for IMU.
Barometer/temp/humidity still need to be taken care of via some kind of SignalK plugin (preferably)

Thanks,
--MG

Tiki26, 2021-06-30:
Well I've Pam's Sean while I had his attention. I've brought his attention
to BBN OS and referenced your OpenPlotter issue. All we can do now is sit
back and wait. I hope you hear from him
Neil

mgrouch, 2021-06-30:
From the commits to pypilot I see that it’s probably done. However this fixes only IMU readings. 

Barometer, temperature should be done in SignalK. It’s a different team independent from pypilot. I do not know if anyone in SignalK team is working on berry IMU v4 support. 

Thanks

Tiki26, 2021-06-30:
Ok
My barometer had always worked using the BM3XXX device but had to apply an
offset. Temperature? No. But I wasn't concerned about temp.
Interestingly, the barometer shows a yellow highlight instead of green when
making the connection to signal K from the I2C app. But the reading is in
there. Not sure if it's me, signal K or the I2C app. It's not bothering me
though

mgrouch, 2021-09-08:
closing this support item not in BBN scope anyway.
If issue persists please reopen it

closed 2021-09-08 by mgrouch