bareboat-necessities/lysmarine_gen #82: No GPS data in open CPN

cornelledepag, 2021-06-13:
OpenCPN doesn't seem to receive gps data the GPS icon is a red dot.

GPS data is available when running the command gpsmon 127.0.0.1:29470 and I get accurate GPS data.

OpenCPN has signalk on port 3000 and signalk_nmea on port 10110 enabled, tried to restart signalk server with no change. 

Getting position from signalk boat finder also doesn't work. I tried to manually add GPS data in signalk with no effect.

Can someone help me out here?

mgrouch, 2021-06-14:
If you get a fix in gpsmon then
you need to check how SignalK is configured to read that from gpsd.
You haven’t mentioned anything 
about your SignalK connection to gpsd. 

Thanks

mgrouch, 2021-06-30:
I’ve seen that in one installation with
ublox gps, when ublox gps is in binary mode instead of NMEA mode. 
gpsd in recent versions resets ublox to binary mode upon start. 

There is a file /etc/default/gps
If I specify -p (—passive) mode
to gpsd then set ublox GPS to NMEA mode and reboot pi everything works fine.

mgrouch, 2021-06-30:
Might be passive mode should be the default for BBN OS

mgrouch, 2021-09-02:
Fixed in BBN OS 2021-09-01

closed 2021-09-02 by mgrouch