bareboat-necessities/lysmarine_gen #90: pypilot support

seandepagnier, 2021-06-29:
There is another service "pypilot_hat" which is optional and powers a SPI lcd as well as if connected a SPI arduino which can receive signals from a wireless remote. 

There are a few other changes to improve pypilot support, and I don't mind making the pull requests to make it easy, but I am wondering if I can just submit to lysmarine and if you intend to merge changes or from that repository or if you have diverged?

mgrouch, 2021-06-29:
Hello Sean,

Thanks for taking a lot at it.

I've sent you an invite with 'write' access.
Few notes:

Every git push will trigger a build in circle CI and if successful new images will be uploaded to cloudsmith https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/?q=lysmarine

There are 2 images need to be tested arm64 and armhf (and armhf can be used in 2 modes 32-bit kernel and 64-bit kernel)

Build duration is about 4.5 hours

Currently I'm out of bandwidth on cloudsmith till ~ july 2, so you will be able to test the images with your changes
only after that date

Thanks,
--MG

seandepagnier, 2021-08-06:
First of all,  I found a few bugs in pypilot which I have fixed.

1)  Should pypilot be running by default?   It seems like it should be enabled if the user decides to, or if pypilot hardware is detected then it could be enabled.   pypilot services do use a few percent of cpu.   different optional services such as pypilot_hat pypilot_web, and option between pypilot, pypilot_boatimu or disabled will vary the cpu usage.

2)  pypilot uses /dev/ttyAMA0 for motor controller.   I did some work on the openplotter-pypilot package which detects this and configures it if needed.   Would it make sense to just integrate the https://github.com/pypilot/openplotter-pypilot repository?   This way, bbn/lysmarine would not need to do as much and the user would have more control of detecting and configuring hardware and services .     This should really be a pypilot-raspberrypi configuration gui for debian.    Do you target other platforms besides raspberry pi?      Are you keeping separate packages of pypilot?   This package automatically can update pypilot, so you would only need to provide this one package.  It depends on your intention for bbn but I feel it may be easier to maintain this way.

3)  I noticed pypilot is run as the user pypilot as a service pypilot@pypilot.   This is great, but shouldn't pypilot_boatimu service also be?

mgrouch, 2021-08-06:
> I noticed pypilot is run as the user pypilot as a service ***@***.*** This is great, but shouldn't pypilot_boatimu service also be?

1. I’d like to have pypilot and pypilot_web started on boot by default. Those who want to disable can easily do so via provided UI service manager application 

2. Openplotter in its current state does more harm than help on BBN. 
If there is configuration app for pypilot it should be part of PyPilot. 
Openplotter individual packages bring base ones and base ones currently break things, apart from issues with permissions in individual packages. Short story Openplotter is not ready for BBN distro. Almost none of packages. 

3. I though boatimu is retired package and pypilot itself can be used instead even just for IMU. I tried it and it worked for me.

Thanks for looking at pypilot in BBN. 
It’s really helpful.

seandepagnier, 2021-08-06:
There is some cpu wasted.  Perhaps pypilot can improve if there are no sensors and use less power.  the web process doesnt use much at all.

The application to detect and configure hardware and set up the services is specific to debian based distros on raspberry pi, so that is why it is not really part of pypilot.

The boatimu is not retired, but a different entry point which allows a smaller footprint to perform imu only tasks.   You are right that you could just use pypilot itself.

I think the most important change is to detect if a pypilot hat is attached which is easy because it creates /proc/device-tree/hat/custom_0 which describes the capabilities of the hat.    If it is found, it should enable the pypilot_hat service and prompt the user if they want to switch the serial ports to enable /dev/ttyAMA0

mgrouch, 2021-08-06:
/dev/ttyAMA0 should be enabled by default in BBN. 

might be pypilot_hat should be started by default too and wait for
/proc/device-tree/hat/custom_0
to appear without failing the service. 

And similarly to pypilot_web it could be disabled in servicedialog 

In this case it’s kind of plug and play.
What do you think? 

Thanks

seandepagnier, 2021-08-07:
Check the port with:
ls -l /dev/serial0

I am not sure you want /dev/ttyAMA0 enabled by default as it disables bluetooth.   There is unfortunately not a good other way to get a reliable serial port on all pis.    The pi4 can use alternate pins for /dev/ttyAMA1-4 but this is so far not utilized by my hardware as the same hardware would not work on other pi which have only the single hardware serial port.

Also. pypilot_hat can work without the custom file if people use their own hardware or my older hardware.       In this case it uses defaults or gets the info from a config file.  The only way to make it plug and play would be to enable it automatically if it detects the file (and it has the contents suggesting a pypilot hat), but may need to be manually enabled if it doesnt.  So making it sleep without the file is not ideal.   Also, this file will only exist if the hat is plugged in at boot, it isn't read later on, so waiting for it doesnt work.  People using other hats will always have a conflict here if pypilot_hat is running since it uses a lot of the gpio pins.

mgrouch, 2021-08-07:
I’m fine with disabling Bluetooth. 
it’s not the protocol I’d rely on. 
On pi4 does it really disable Bluetooth?

ls -l /dev/serial0 is pointing to /dev/ttyS0 in current default BBN installation. 

I’ll check .service systemd syntax to see if pypilot_hat can be launched conditionally based on presence of that file or based on result of some arbitrary script. 

I don’t think pi OS recognizes any hats. I might be wrong. All other hats have to be configured by the user one way or another by loading more drivers. And obviously if hardware is put together the way it has conflicts it can’t be supported anyway by the software. It’s up to the user to put together hardware configuration which can work. 
Just need to cover most common hardware combinations.

seandepagnier, 2021-08-07:
I think it does bluetooth disable even on the pi4 but I do not use bluetooth to verify it.

Yes, an arbitrary script would likely be best.    The os can recognize hat via an eeprom on i2c 0.   This eeprom can contain device descriptors as well as custom files.   pypilot relies on the file while the pypilot_hat service reads to determine what is connected to the io pins.   If this file is missing it assumes the default configuration which is a nokia5110 screen.

mgrouch, 2021-08-07:
Conditions and Asserts in Systemd Unit files:

https://www.freedesktop.org/software/systemd/man/systemd.unit.html#Conditions%20and%20Asserts

mgrouch, 2021-09-08:
closing this
if there are other issues report it separately

closed 2021-09-08 by mgrouch