bareboat-necessities/lysmarine_gen #100: Add support for local languages via language packs

mgrouch, 2021-07-31:
Add support for local languages via language packs

https://github.com/bareboat-necessities/bbn-os-language-packs

Bazulo, 2021-12-09:
I can help with French language packs. 
I'm already on Crowdin, if that's the tool you're using.

mgrouch, 2021-12-09:
great! I’m not using crowdin. 
if you have an experience in localizations of the software I’d appreciate some guidance from you how we can approach it. 

Are you a user of BBN OS? If you are you probably know better areas of priority where localization is needed. 

So better start with those. 

If you browse source code 
I think places to start are all
.desktop files
constants.js under bbn-launcher
and few scripts 
servicedialog.sh
bbn-commands.sh

SignalK I think is missing localization but it’s a separate project. 

Thanks

mgrouch, 2021-12-09:
Oh. It would be nice to translate the whole documentation too. 

The file to be edited for that (single file)

https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/docs/bareboat-os.adoc

Bazulo, 2021-12-09:
Hi,

I do have some experience with localizing software and websites. Enough to say that it is a daunting task and it should definitely be approached with a translation platform like Crowdin, or some other one which also has an interface with GitHub. In a nutshell, every label, message, echo, log or any word that appear or could be appearing on or from the user interface has to be made a variable dynamically accessed according to a language specification. Those text elements are stored in external files that are used by the program as well as by the translating platform. Twenty years ago, I was hardcoding the translations within the program. It gets messy quite rapidly and you can’t see your code straight after a while. And if you want more than one translation, it becomes exponentially more bloated and difficult. So, there is a need to plan ahead and modify your code architecture to facilitate localization in multiple languages. The bad news is I don’t know of any free platform for that purpose. The good news is that once it is set up for one language, adding other translations is quite easy.

The documentation is easier to manage. You basically replicate the generating script with the text parts translated. I have some experience with LaTex, yours seems to be AsciiDoc, right? I could start with that, but then again it should make reference to the user interface, which would still be in English.

As a very first step before going into translation per se, it is wise to review the original text in view of it being translated. Simple sentences, simple words, etc. Maybe I could  start with that, reviewing the documentation so it will be more straightforward to translate?

On another note, I was talking this morning with Peter Antypas, the maker of MAIANA (https://github.com/peterantypas/maiana <https://github.com/peterantypas/maiana>) who tells me that he would very much like to see the MAIANA well integrated within BBN OS. As you may know, OpenPlotter just did that (I recently translated the language pack to that effect) and has added his product to their storefront (https://shop.openmarine.net/home/15-maiana-ais-base-kit.html <https://shop.openmarine.net/home/15-maiana-ais-base-kit.html>).

Here is his email address: ***@***.*** ***@***.***>

Fair winds,

Denis

mgrouch, 2021-12-09:
@Bazulo 

The documentation in the form the user would see it is here 

https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html

I’d be glad to include Maiana software. I do not have the hardware to test it on BBN OS so I think it’s better if Mariana author could put code for maiana configuration utility as a part of maiana GitHub project. And we can work on packaging it into BBN OS.

mgrouch, 2022-08-04:
I've registered a project on Crowdin

https://crwd.in/bareboat-necessities-bbn-os

Contributions to translations are welcome

Thanks

mgrouch, 2022-08-04:
With the word count in the docs crowdin would cover only 4 languages on the free plan.

The following languages translations needed:

- Spanish
- French
- Dutch
- Portuguese 

Thanks

Bazulo, 2023-04-16:
Hi Mikhail,

We were briefly in contact last year and the year before about translation in other languages of BBN OS. Serious health issues prevented me from following up, but it seems I’m out of the woods now. So I’m circling back, reviving former implications in interesting projects.

I hope yours is still alive and kicking. I still have a great interest in it as you have many interesting a novel aspects in your approach to boat ownership/handling/living. Copious and relevant documentation being chief among them. 

Automatic translators are becoming very efficient, to the point that relying on them will become, I believe, an acceptable and standard practice. This means that instead of producing a different text for every language, you produce one text in the original language - but it has to be tailored to be easily used by automatic translators. 

Take for example this first paragraph of your documentation:

"Why use BBN OS? Some people use just commercial marine electronics on board, some ones use smart tablets and marine Wi-Fi multiplexers. All these solutions lack one important component - an onboard computer capable of monitoring and controlling all aspects of the boat systems, and capable of storing data. Some people put an Intel-based computer for it. Intel-based solutions are more power consuming. With BBN OS you can build a central boat computer meeting your needs. All on low power consuming raspberry pi with flexibility of adding countless choices of sensors to talk to all boat systems, Internet, local Wi-Fi, cameras, NMEA network, and a system able to decode marine radio protocols. Initially this project started from experiences of using OpenPlotter to improve user experience. BBN OS is free and open source. It is based on commonly used community supported open source projects such as SignalK, PyPilot, OpenCPN, and others. BBN OS graphical user interface will let you build a cockpit front-end to all functionality of the OS from chartplotting, dashboards, weather information, to media player, etc."

This leads to very poor automated French translation. You still can understand most of it, if you are not discouraged after a few sentences, but it is quite unpalatable. But if you replace it by this one:

"BBN OS is an operating system for an on-board central computer capable of collecting, storing and managing data as well as monitoring and controlling all aspects of a boat systems. Made for Raspberry Pi, BBN OS, despite having very low power requirements, has the flexibility to add countless sensors, the Internet, a local Wi-Fi, cameras, NMEA networks and radio protocols. It is free, open-source and based on many community projects such as Signal K, PyPilot, OpenCPN and others. BBN OS will also let you build a cockpit graphical user interface to all functionalities: chartplotting, dashboards, weather information, media player, etc. »

This gives a very good translation in French, I would dare a native French speaker to tell if it was automatically made or not. Of course the original English language has to be simpler, but I think it even gains clarity in English, making it all the while more accessible. And my guess is that once it’s done and made efficient for French, automatic translation for other languages should be of very good quality and it should afterwards be easier to find native speakers of other languages fluent in English that could suggest adjustments in the original text to improve automated translation in their language without affecting the automated translation in other languages.

Rewriting the original text for automated translation is quite a task. But I’m willing to tackle it if you find this approach worthwhile and don’t mind your original text being transformed and sometimes completely rewritten. What do you think?

Fair winds,

Denis

mgrouch, 2023-04-16:
Hello Denis,

I’m against editing original English. The fact your edited piece translated fine into French doesn’t mean it’s going to translate automatically fine into some other language. 
I’m fluent in Russian and it might be shows in my English way of building sentences so my English translates automatically fine into Russian. It might be the case for your French translation.  Few people joined croudin and translate it there. There are some parts already translated into German. So I suggest you to join croudin and contribute your French translation there. It’s the right tool for that job. 

Thanks!

mgrouch, 2023-04-16:
@Bazulo 

Crowdin link

https://crwd.in/bareboat-necessities-bbn-os

Jade64, 2023-10-21:
Dear @mgrouch.

Above you listed 4 languages for the free plan. I am German and open to support translation. If some day there is a german version, then it would not be free?

What about the overall liveness of the project? It look feature complete and stable. Is this the reason for github activities are quite low?

BR Jens

mgrouch, 2023-10-21:
You can switch to German in already released BBN version. This thread is about documentation translations. 

Yes GitHub activity is low due to maturity and stability of the system.

closed 2023-12-19 by mgrouch