bareboat-necessities/lysmarine_gen #120: Bluetooth turned off

Kaythaltes, 2021-09-24:
I'm not able to turn on Bluetooth. When I try in settings I can change the switch but it keeps saying 'Bluetooth turned off and no devices are shown. When I hoover the mouse over the Bluetooth sing in the top menu it shows 'Bluetooth active'
[image]

ericrouah, 2021-09-24:
This is same for me.

mgrouch, 2021-09-24:
Might be output of these could shed come light:

as root:

rfkill

```
 dmesg | grep hci0

 systemctl status bluetooth

 hciconfig -a

```

Kaythaltes, 2021-09-24:
[image]

mgrouch, 2021-09-24:
Reported 

Here: https://github.com/raspberrypi/linux/issues/4601

and here: https://github.com/raspberrypi/linux/issues/4239

mgrouch, 2021-09-24:
couple more:

https://github.com/raspberrypi/firmware/issues/1150

https://github.com/raspberrypi/firmware/issues/1539

@ericrouah 
@Kaythaltes 

Is anything else connected to your raspberry pi? Any hats? Any PCBs? SSD boards?
Anything that uses UART of raspberry pi?

ericrouah, 2021-09-24:
Nothing yet connected to a bare Raspberry Pi 4/8GB

mgrouch, 2021-09-24:
@ericrouah 

> Nothing yet connected to a bare Raspberry Pi 4/8GB

If you comment out

arm_64bit=1

in /boot/config.txt

do you still have that issue after reboot?

thanks

ericrouah, 2021-09-24:
this is the results 👍 
user@lysmarine:/var/www/bbn-launcher $ cd /..
user@lysmarine:/ $ bluetoothctl
Agent registered
[bluetooth]# power on
Failed to set power on: org.bluez.Error.Failed

ericrouah, 2021-09-24:
> @ericrouah
> 
> > Nothing yet connected to a bare Raspberry Pi 4/8GB
> 
> If you comment out
> 
> arm_64bit=1
> 
> in /boot/config.txt
> 
> do you still have that issue after reboot?
> 
> thanks

I done this and nothing changed

ericrouah, 2021-09-24:
after uncomment arm_64bit=1 and reboot
I get 
user@lysmarine:/var/www/bbn-launcher $ cd /..
user@lysmarine:/ $ bluetoothctl
Agent registered
[bluetooth]# power on
Failed to set power on: org.bluez.Error.Failed
[bluetooth]# 

@mgrouch this looks as same

ericrouah, 2021-09-24:
if I test

user@lysmarine:/ $ /usr/bin/btattach -B/dev/ttyAMA0 -Pbcm -S115200
Attaching Primary controller to /dev/ttyAMA0
Failed to open serial port: Device or resource busy
No controller attached
user@lysmarine:/ $

ericrouah, 2021-09-24:
@Kaythaltes I have same results in the request done like this one 👍  https://github.com/bareboat-necessities/lysmarine_gen/issues/120#issuecomment-926754865

mgrouch, 2021-09-24:
I’ve narrowed this down. 
If I turn off pypilot services
then Bluetooth works. 
now question what is pypilot doing
to break it

mgrouch, 2021-09-24:
Report filed here now 

https://github.com/pypilot/pypilot/issues/108

mgrouch, 2021-09-24:
For a fix try

sudo sed -i 's/ttyAMA0/serial0/' "$(find /usr/local/lib -name serialprobe.py)"

and reboot

mgrouch, 2021-09-24:
fixed in main branch of BBN OS on github

ericrouah, 2021-09-24:
AFter I done that, the result is :

***@***.***:/ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled;
vendor preset: enabled)
   Active: active (running) since Sat 2021-09-25 00:14:51 BST; 2min 26s ago
     Docs: man:bluetoothd(8)
 Main PID: 1125 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─1125 /usr/lib/bluetooth/bluetoothd

Sep 25 00:14:51 lysmarine systemd[1]: Started Bluetooth service.
Sep 25 00:14:51 lysmarine bluetoothd[1125]: Starting SDP server
Sep 25 00:14:51 lysmarine bluetoothd[1125]: Bluetooth management
interface 1.18 initialized
Sep 25 00:14:51 lysmarine bluetoothd[1125]: Sap driver initialization failed.
Sep 25 00:14:51 lysmarine bluetoothd[1125]: sap-server: Operation not
permitted (1)
Sep 25 00:14:51 lysmarine bluetoothd[1125]: Failed to set privacy:
Rejected (0x0b)
Sep 25 00:15:17 lysmarine bluetoothd[1125]: Endpoint registered:
sender=:1.35 path=/MediaEndpoint/A2DPSource
Sep 25 00:15:17 lysmarine bluetoothd[1125]: Endpoint registered:
sender=:1.35 path=/MediaEndpoint/A2DPSink
Sep 25 00:15:24 lysmarine bluetoothd[1125]: Failed to set mode: Failed (0x03)
Sep 25 00:15:41 lysmarine bluetoothd[1125]: Failed to set mode: Failed (0x03)
***@***.***:/ $

mgrouch, 2021-09-24:
You need to run second line too.
I’ve edited my post since I first posted it. 

It fixes the issue. sorry I was too fast and didn’t post correct fix initially.

ericrouah, 2021-09-24:
Yes I used the two command lines. And the result is same.

ericrouah, 2021-09-24:
user@lysmarine:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-09-25 00:38:40 BST; 6min ago
     Docs: man:bluetoothd(8)
 Main PID: 1122 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─1122 /usr/lib/bluetooth/bluetoothd

Sep 25 00:38:40 lysmarine bluetoothd[1122]: Bluetooth management interface 1.18 initialized
Sep 25 00:38:40 lysmarine bluetoothd[1122]: Sap driver initialization failed.
Sep 25 00:38:40 lysmarine bluetoothd[1122]: sap-server: Operation not permitted (1)
Sep 25 00:38:40 lysmarine bluetoothd[1122]: Failed to set privacy: Rejected (0x0b)
Sep 25 00:39:06 lysmarine bluetoothd[1122]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource
Sep 25 00:39:06 lysmarine bluetoothd[1122]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink
Sep 25 00:39:19 lysmarine bluetoothd[1122]: Failed to set mode: Failed (0x03)
Sep 25 00:39:31 lysmarine bluetoothd[1122]: Failed to set mode: Failed (0x03)
Sep 25 00:40:03 lysmarine bluetoothd[1122]: Failed to set mode: Failed (0x03)
Sep 25 00:42:57 lysmarine bluetoothd[1122]: Failed to set mode: Failed (0x03)
user@lysmarine:~ $

mgrouch, 2021-09-24:
Can you post command lines you used?
Thanks

mgrouch, 2021-09-24:
Or better at this point the whole file  serialprobe.py

mgrouch, 2021-09-24:
Did you run these ???

```
sudo sed -i 's/ttyAMA0/serial1/' "$(find /usr/local/lib -name serialprobe.py)"
sudo sed -i "s/'ttyAMA'//" "$(find /usr/local/lib -name serialprobe.py)"
```

mgrouch, 2021-09-25:
In your serialprobe.py

Line 330 needs to look like this:
`
if device == '/dev/serial1' and name != 'servo':`

and  line  71 like this:

`     devicesp = []`

and its a fix I've tested already on arm64 kernel and arm32

Thanks

mgrouch, 2021-09-25:
ok. I did more testing.
What I see sometimes it works sometimes it doesn't
It looks like pypilot code still finding and probing bluetooth device instead of serial port.
The probing code also goes via different paths under /dev/ filesystem.

At this instead of my quick fix I think it's better if it is done properly in pypilot code.

Thanks for reporting this. Without it the issue would still be hanging.
Do you even need pypilot for your use case?

Thanks

Kaythaltes, 2021-09-25:
Problem seems to be solved with this fix.
Thanks

mgrouch, 2021-09-25:
@ericrouah 
@Kaythaltes 

Better fix is committed into main of https://github.com/bareboat-necessities/lysmarine_gen/blob/master/install-scripts/4-server/files/blacklist_serial_ports

Only installing blacklist_serial_ports in required after all.

into /home/pypilot/.pypilot/

ericrouah, 2021-09-25:
I am sorry but I cannot do that. The folder /home/pypilot/.pypilot/ is owned by --system (pypilot) and as a user I can only read it.

mgrouch, 2021-09-25:
@ericrouah 

sudo su
su - pypilot 

That will make you a pypilot user.

ericrouah, 2021-09-25:
and yes it works !!!
Thank you for your patience ...

Kaythaltes, 2021-09-27:
Sorry but I don't have too much experience with Linux.
How to install this file?

mgrouch, 2021-09-27:
> Sorry but I don't have too much experience with Linux. How to install this file? Op za 25 sep. 2021 19:08 schreef Mikhail Grushinskiy < ***@***.***>:
> […](#)

By install I meant just copy.

closed 2021-10-04 by mgrouch