bareboat-necessities/lysmarine_gen #142: add support for rpi4 alternatives

peronosporos, 2021-12-26:
rock 3b eg. that is to be released soon will feature in contrast to the pi: lower idle consumption and higher efficiency, better heat management, full hdmi, poe, power on/off button, rtc, 3.5mm jack with mic, wifi 6, native support for m2 4g/5g modem, emmc, camera, m2 ssd, it has more convenient port layout, etc.

but also rock 3a, rock pi, etc are interesting options as well. drivers open sourced.

mgrouch, 2021-12-26:
I do not have rock hardware. 
BBN OS is tested on rpi4, CM4 and rpi400. I do not know any single user of BBN OS who has the hardware you mentioned above or planning to have it.

peronosporos, 2021-12-26:
i am considering to make it with such hardware, as i think it would be more suitable than a pi for the intended use. is it out of the intentions of this project to support non-rpi hardware?

mgrouch, 2021-12-26:
in order to claim that it works on other hardware it needs to be tested on other hardware. So someone has to have that hardware and test on it

mgrouch, 2021-12-28:
I’m going to close this. I just do not see enough actual user base that would make it worth the effort.

closed 2021-12-28 by mgrouch

pandaero, 2023-05-04:
I would like to see if this is doable, I got a RockPi 4C Plus to try this on (in the next couple of weeks), any tips or considerations? I am thinking just to prepare an image with balenaEtcher and fire it up

pandaero, 2023-05-04:
After checking out the [RockPi OS List](https://wiki.radxa.com/Rockpi4/downloads), maybe the approach of a Debian distribution and installing BBNOS components is better.

mgrouch, 2023-05-04:
Pick a base headless Debian based image for rockpi
Fork BBN scripts code on GitHub 
Create circle CI account
Modify scripts in your fork for rockpi
Run builds on CircleCi 
Upload resulting build somewhere.  ( I use cloudsmith )

Repeat the process until you happy with
fully working image