bareboat-necessities/lysmarine_gen #143: add support for sailor hat

peronosporos, 2021-12-26:
sailor hat adds nmea 2000 connectivity with option to be used as power supply at the same time, power supply at 12/24v, battery. firmware is foss. costs 64€.

https://github.com/hatlabs/sh-rpi

https://hatlabs.github.io/sh-rpi

mgrouch, 2021-12-26:
Sorry I do not have that hardware even to test.  
It’s again up to vendor to support it. 
If vendor supports raspberry pi OS then same methods would work on BBN OS.

mgrouch, 2021-12-28:
I suggest you opening this as an issue in sailor hat git project.

mgrouch, 2022-01-07:
@peronosporos 

Anything else is required apart from running install script mentioned here

https://github.com/hatlabs/SH-RPi-daemon

?

Thanks

mgrouch, 2022-01-21:
Added via add-ons in buster branch. Should make it into next release.

closed 2022-01-21 by mgrouch