bareboat-necessities/lysmarine_gen #147: package installer/uninstaller

peronosporos, 2021-12-26:
consider maybe a gui installer/uninstaller of featured packages to avoid getting overwhelmed by all the software at once, reduce installing time, give better control of the system, reduce failure points, and have a smaller disk occupation according to needs

mgrouch, 2021-12-26:
There is apt which is package installer

peronosporos, 2021-12-26:
no, i mean to avoid a pre-installed mountain of software and make the user more involved into which packages are installed. something like how openplotter manages software with a "settings" application that has nothing pre-installed and everything is up to the user to implement

mgrouch, 2021-12-26:
that makes it virtually untestable. 
Openplotter will install anything that it available at the time and what wasn’t tested properly. No thanks.

closed 2021-12-27 by mgrouch