bareboat-necessities/lysmarine_gen #150: HF marine info organisation

peronosporos, 2021-12-26:
consider making a database for weatherfax and navtex frequencies and timeschedules, and using gps and local time to use a script to automate the process of receiving and decoding the info both with rtl-sdr if it is the only option, and ssb if an installation exists already

mgrouch, 2021-12-26:
Shouldn’t be difficult to do on condition of having hardware to test it properly

mgrouch, 2022-02-17:
WeatherFax schedules are installed with opencpn weatherfax plugin

mgrouch, 2022-02-17:
NOAA satellite weather 

Some links:

https://github.com/jekhokie/raspberry-noaa-v2

https://www.rtl-sdr.com/rtl-sdr-tutorial-receiving-noaa-weather-satellite-images/

https://www.rtl-sdr.com/an-alternative-noaa-weather-satellite-tutorial-using-rtl_fm-and-wxtoimg/

https://www.rtl-sdr.com/raspberry-noaa-v2-raspberry-pi-automated-noaa-and-meteor-weather-satellite-capture/

https://www.rtl-sdr.com/creating-an-automated-raspberry-pi-and-rtl-sdr-based-noaa-weather-satellite-station/

https://www.rtl-sdr.com/automatically-receiving-decoding-and-tweeting-noaa-weather-satellite-images-with-a-raspberry-pi-and-rtl-sdr/

https://www.rtl-sdr.com/youtube-tutorial-software-set-up-for-a-portable-raspberry-pi-and-rtl-sdr-based-noaa-weather-satellite-receiver/

mgrouch, 2022-02-17:
Inmarsat-STD-C decoding with SDR:

https://www.rtl-sdr.com/rtl-sdr-tutorial-decoding-inmarsat-std-c-egc-messages/

https://github.com/cropinghigh/stdcdec

mgrouch, 2022-02-17:
NavTex decoding with SDR

https://www.arachnoid.com/JNX/

https://www.rtl-sdr.com/decoding-navtex-with-an-sdrplay-rspdx-sdruno-and-yand/

mgrouch, 2022-02-17:
Requesting and receiving GRIB files via SSB radio

Grib Plugin in OpenCPN

WinLink PAT  https://github.com/la5nta/pat

jpskmail  https://github.com/bareboat-necessities/javapskmail-VK2ETA