bareboat-necessities/lysmarine_gen #161: Add support for more I2C sensors / I2C sensor management console.

deanfourie1, 2022-01-24:
Please can we see more supported I2C sensors in OpenCPN. There is such limited options out there and what for?

Also, some kind of I2C management console.

mgrouch, 2022-01-24:
Realistically you do not want to connect many i2c sensors to pi directly. 

Esp32 controllers are using less power and can run and provide data to your network even if pi is off. 

So you would connect your environmental i2c sensors and others voltage, power
via SensESP or esphome.io

deanfourie1, 2022-01-24:
I am trying to avoid wifi as much as possible. So with this theory, use the esp32 then to convert the I2C Serial data to nmea?

mgrouch, 2022-01-24:
SensESP sends data to SignalK using SignalK protocol. From SignalK you can convert to NMEA. It allows using esp32 via WiFi and USB serial. Might be Ethernet as well but you need to have your board with Ethernet. 

esphome integrates with Homeassistant and supports more variety of sensors. There is no ready to use integration with NMEA or SignalK. However I’ve seen some people feeding SignalK from homeassistant.

mgrouch, 2022-03-05:
Seems easy to set if the sensor is supported.
sudo i2cdetect -y 1

and pretty much all is needed for setup is knowing i2c address for the device

closed 2022-03-05 by mgrouch

mgrouch, 2022-10-18:
Some kind of server written in rust on top of 

https://github.com/golemparts/rppal/

https://github.com/rust-embedded/embedded-hal

would allow many sensors compatible with Rust embedded-hal

https://crates.io/search?q=embedded-hal

mgrouch, 2024-06-13:
I2C scanner in rust

https://github.com/bareboat-necessities/lysmarine_gen/discussions/420