bareboat-necessities/lysmarine_gen #169: Stuck on "Wait for desktop to start"

emilecantin, 2022-02-15:
So I've tried the latest build, and it just dumps me to a command-line with the Lysmarine logo and the sentence "Wait for desktop to start" as a welcome message.

Do I need to do anything more than this? The instructions hind at waiting a few minutes, but it's now been well over half an hour of this. It's really not clear if there's anything else to do, or if it's a bug.

I'm running this on a Raspberry Pi 3, by the way.

mgrouch, 2022-02-15:
Pi3 is not supported per system requirements:

https://github.com/bareboat-necessities/lysmarine_gen#system-requirements

It needs pi4, CM4 with WiFi, or pi400

closed 2022-02-15 by emilecantin

mgrouch, 2022-02-16:
Running chartplotter which displays charts requires better hardware than pi3.