bareboat-necessities/lysmarine_gen #174: bbn for rpi 4 : stuck on waiting for desktop to start. then black screen

jimmid709, 2022-04-30:
Had BBN running on my rpi 4 then installed gparted.
it said to run gparted type sudo startx

since then no desktop when start the raspberry pi 
how to fix?

mgrouch, 2022-04-30:
Since you were messing around with partitioning I’d say burn a new BBN OS image and reinstall from scratch

jimmid709, 2022-04-30:
I only installed gparted but didn't use it. I was going to format a usb but didn't.
In the start up terminal it did come up with BBN and something else on the command line, but when I clicked enter it said file not found.
What is the command to load the desktop as I have configured it a bit and would loose all that with a new copy.

mgrouch, 2022-04-30:
startx is the command to start desktop. 
however you seems somehow lost some files if you are getting file not found errors. Do I’d recommend full reinstall

closed 2022-05-01 by mgrouch