bareboat-necessities/lysmarine_gen #175: pypilot_hat support.

seandepagnier, 2022-05-24:
It seems with latest version, the pypilot_hat script fails.   The dependencies are not installed correctly.  I am not sure exactly why but maybe:
    git checkout a820bcda00f12c37cba8294b1f8ea0206a84b429 # Jan 10, 2022

I think it is safe to remove this line..  I fixed the problem by doing a clone and install of pypilot, so I think that commit is broken.

Also,  the dependencies script gets executed from the setup.py script by default and handles a lot of the dependencies in the pypilot install script, including pypilot_data

mgrouch, 2022-05-24:
How can’t I check if pypilot_hat script is working? I do not have pypilot hat. 
I’ll try to make changes for the next build

seandepagnier, 2022-07-29:
Hi,  there are a few integration issues:

1)   the pypilot user has a blacklisted_serial_ports file in pypilot folder disabling ttyAMA0 which is most commonly used.
2)   how to enable/disable the pypilot_hat service?
3)   pypilot needs /dev/ttyAMA0 by dtoverlay=disable-bt   This may be undesirable for non-pypilot users who want bluetooth.

It seems services are not systemd so you cannot use systemctl?   Ok, how do I enable disable services from command line in bbn?

pypilot services do use some cpu.    I suggested as before disabling by default, but at boot check for /proc/device-tree/hat/custom_0 and the contents will indicate what hardware is available for pypilot.   This could then enable the pypilot services automatically and configure the hardware serial (or this could be tied to enabling pypilot?)  so users with this  pypilot hardware would have zero configuration while other users would not incur any cpu/resource penalty.     Can you point me to where to put such a script in your repository?

mgrouch, 2022-07-29:
Default BBN config is bluetooth enabled for music, etc. Pypilot enabled for IMU functionality only.
This is most common use case.

Enabling pypilot motor controller and hat is documented for those who need actual autopilot. See:

https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_pypilot

it involves loosing bluetooth functionality and removing ports from blacklisting.

This procedure has been tested on actual installation on a boat.

PyPilot scripts are here:

https://github.com/bareboat-necessities/lysmarine_gen/blob/buster/install-scripts/4-server/files/

install script is here:
https://github.com/bareboat-necessities/lysmarine_gen/blob/buster/install-scripts/4-server/02-pypilot.sh

All services use systemctl. You use systemctl to enable and disable services in BBN. 
Take a closer look at the doc https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html

PyPilot uses negligible CPU in my tests.

On installation where we tested it  /proc/device-tree/hat/custom_0 wasn't there even with pypilot hat attached.
User ended up removing hat completely and using just motor controller and web page on phone to control pypilot.

How are you going to handle automatically
dtoverlay=disable-bt

only pypilot users with motor controller need it. Others need bluetooth

Thanks

seandepagnier, 2022-07-29:
I understand by default it should not have bluetooth disabled.

If you use the gui and navigate to services, there is a pypilot@pypilot service and a pypilot_web service but no pypilot_hat service even though it is running.    How is this?

The system you tested used very old hardware and did not have the eeprom attached.   So for that user they would have to do the manual setup as now.

For pypilot more recent (last 2 years or so)  the custom_0 file does exist.    It will only exist if the hat is present at boot, not plugged on later.   Anyway,   If this file does exist and it indicates pypilot, then I suggest it automatically modify the boot to disable bluetooth, as well as enable pypilot services (if not by default) and disable blacklisting any ports.    If the file is not detected or it does not have contents indicating a pypilot hat, it would work exactly as it does now.

Would this be acceptable, and where would you suggest adding this helper logic?

mgrouch, 2022-07-29:
We have to support old hardware too. That's why I commented out conditional starting of pypilot_hat service
based on custom_0  presence.

Service UI doesn't have to expose all services. Just commonly used ones sometimes needed to be disabled or enabled.

For configuration users have means and documentation to enable pypilot_hat service from command line.

Unfortunately /boot/config.txt doesn't allow conditional logic based on some file presence.
And the line which disables bluetooth is in it.

mgrouch, 2022-07-29:
Thinking about it. I'm fine with it. If you modify  /boot/config.txt on first boot based on custom_0 presence and making sure it belongs to pypilot_hat. and enable pypilot_hat service.

I'm fine with proposed changes. At least they will simply setup for people with new pypilot_hat hardware.

mgrouch, 2022-07-29:
you can commit into buster branch directly. I sent you invite a while back. No PR needed.
Thanks

mgrouch, 2022-07-29:
This logic can be invoked as a script from a service which gets run on start-up and goes away.
There are plenty of services like that

mgrouch, 2022-07-29:
Side effect is that people would need to reboot again after /boot/config.txt change. But after that they will be fine.

seandepagnier, 2022-07-30:
Ok we are on the same page.   Which script do you recommend I edit?  You said "plenty of services"   so which one?  rc.local?

mgrouch, 2022-07-30:
no. New service for systemd. 
pypilot-detect.service

so everyone knows it’s for pypilot

mgrouch, 2022-07-30:
Ok. Thanks. I’ll build it in few days. It takes longer than cooking a turkey. One script to run. 
I’ll try to find what other improvements I can squeeze into next one.

mgrouch, 2022-07-30:
I do not see any commits from you. 
Did you push it into buster branch?

mgrouch, 2022-07-30:
Sorry I've misread your other comment on separate issue as "Try another image" and I mistakenly decided that you already committed changes.

seandepagnier, 2022-08-01:
Sorry I committed to the master branch..

before I correct anything can you explain the master vs buster branches?

mgrouch, 2022-08-02:
Master is abandoned branch. All developments and releases happen from buster branch. 
there are some other branches. They were basically for various experiments. 

Buster branch is default and current.

mgrouch, 2022-08-02:
I've merged your changes with some corrections of mine into buster branch.
Please review as I will not have hardware to test.
Thanks

mgrouch, 2022-08-02:
Also I have concerns about update_boot_conf function.
It just does grep for a string and what if that string is just inside the comment?

mgrouch, 2022-08-02:
I've rewrote it as shell script pypilot_detect.sh

seandepagnier, 2022-08-02:
I dont think it does the same thing anymore.   What is the advantage of using shell instead of python here?

1)   it does not parse the json anymore, it is just grepping for 'mpu' etc without verifying it is in json format.   This is much more likely to get a false positive with some other eeprom/hat.    Would be hard to extend in the future..  even the gpio port for lirc is designated in the json format so should be parsed (not implemented in the script yet...)
2)   Does not enable the hat service anymore (but does remove the blacklist I forgot about),  does not enable lirc service
3)   If the user comments out the line added dtoverlay etc, it will now not honor this, and instead add another line overriding it.   It was meant to match if commented out.   It was supposed to match in a comment.
4)   Rather than use a function call..  the same logic is repeated 3 times...

Thanks for this, just wanted to clear this up.

mgrouch, 2022-08-03:
I think some of the functionality is generic and should be in pypilot code base. It's not BBN specific
So I created proposal here

https://github.com/pypilot/pypilot/issues/142

Thanks

mgrouch, 2022-09-04:
The latest BBN Marine OS based on Bullseye needs more testing before  this can be closed

mgrouch, 2022-09-09:
initial support added

closed 2022-09-09 by mgrouch