bareboat-necessities/lysmarine_gen #190: Waveshare 10 DOF IMU Sensor (D) not showing in PyPilot

samdowrick, 2022-07-22:
I have the Waveshare 10 DOF IMU Sensor (D) which has the ICM20948 IMU. I have followed the guide on your website but nothing is showing up on PyPilot. The BMP280 is showing in signalK and i can see the address of the IMU within terminal but PyPilot is not showing any data.

mgrouch, 2022-07-22:
Thanks for the report. It's good to know that version (D) doesn't work.
I have Waveshare 10 DOF IMU v2 (C) and that one works just fine.
They do have different chipsets

samdowrick, 2022-07-22:
I take it PyPilot doesn't know what it's looking for then? Is there anyway to add a definition to a config file somewhere so that it would work with this sensor?

mgrouch, 2022-07-22:
It would be in this code: https://github.com/seandepagnier/RTIMULib2

samdowrick, 2022-07-22:
That seems to include a IMU driver for the ICM20948 that is featured on the version D board so im not too sure.

mgrouch, 2022-07-22:
https://www.waveshare.com/wiki/10_DOF_IMU_Sensor_(D)

They do provide code link in that wiki page.
So it might be a matter of looking at that code and doing same in RTIMULib2.
Might be in auto-detection part

samdowrick, 2022-07-24:
Got it working, was an address conflict woth the rtc

mgrouch, 2022-07-24:
You didn’t mention before that you have RTC installed. Glad everything works

closed 2022-07-24 by mgrouch

samdowrick, 2022-07-25:
I will point out that the rtc (waveshare rtc watchdog) has no water of changing the i2c address easily, and neither does the imu without removing chios and soldering jumpers.

mgrouch, 2022-07-25:
There is a not it the documentation for RTC

https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_rtc_real_time_clock

to watch out for i2c conflicts