bareboat-necessities/lysmarine_gen #194: Pican-m not working

samdowrick, 2022-07-24:
Followed all the steps in the pican-morning manual however it gets stuck at one point. This command - sudo /sbin/ip link set can0 up type can bitrate 250000 - does nothing, just shows a blank screen in terminal. Any reason why this is happening and how can I fix it?

mgrouch, 2022-07-24:
Where is pican-M manual? Links pls?
What is expected output of you command?

samdowrick, 2022-07-24:
https://copperhilltech.com/content/pican-m_UGB_20.pdf

mgrouch, 2022-07-24:
So ip command is not supposed to produce any output. 
output should come from candump command

mgrouch, 2022-07-25:
I've added more documentation for NMEA 2000

https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_nmea_2000

It should make it more clear.

samdowrick, 2022-07-25:
You're documentation works. In fact I've got so much data flowing in that opencpn keeps freezing. Time to remove some unused software.

closed 2022-07-25 by samdowrick

mgrouch, 2022-07-25:
Unused software is not running so it wouldn’t cause OpenCPN freezes. 
You should report OpenCPN freezes via OpenCPN GitHub issues.