bareboat-necessities/lysmarine_gen #199: boot speed

seandepagnier, 2022-07-29:
I noticed it takes many minutes to boot...    Is this normal?   It is much slower than other distributions.   I did not really look closely as to why but I am wondering if there are any plans to optimize startup?   It is likely one or two things that could be deferred?

Otherwise, what about waking into a saved hibernation state to speed startup?

mgrouch, 2022-07-29:
For me it boots in 35sec for me from the moment I press power on button till I see mouse pointer in GUI.

Never seen it in many minutes. Do you have bad SD card or SSD?

During install it would take longer on first two boots as it compiles some additional stuff.

I suggest users in documentation to use SSD instead of SD card for final on boat install.

I think storing hibernation state might be unnecessary complication.

Thanks

mgrouch, 2022-07-29:
I've rebooted many different users installations over VNC cloud connection and I've always seen 
even desktop coming back in VNC in under a minute. There is definitely some issue (hardware?, SD card) in your set up.

Thanks

seandepagnier, 2022-07-29:
Does it require internet access to complete the two initial slow boots?

mgrouch, 2022-07-29:
@seandepagnier 
It doesn’t require internet access

seandepagnier, 2022-07-30:
ok,   I will try a fresh image.

mgrouch, 2022-08-02:
I do not see issues with boot speed. On all installations I did it's fine and inline with other rpi distributions

mgrouch, 2022-08-22:
I can see that it is an issue now. I've committed fixes now into bullseye branch which is default.
I'll be able to build a new image early September (I've run out of cloudsmith quotas for this month)
Thanks

mgrouch, 2022-08-22:
https://github.com/bareboat-necessities/lysmarine_gen/issues/220

mgrouch, 2022-09-06:
BBN Marine OS 2022-09-06 has a fix for this issue

closed 2022-09-06 by mgrouch