bareboat-necessities/lysmarine_gen #207: Wifi Configuration from headless is difficult

martincaddick, 2022-08-10:
I ended up plugging in a cable, setting the fixed resolution, using vnc and then connecting that way. would be so much better if it was scripted ;-)

mgrouch, 2022-08-10:
Not sure what was difficult. WiFi AP is preconfigured on the image. There is nothing to configure except of changing the password.

martincaddick, 2022-08-11:
Apologies I should have been more specific. I was trying to have it connect to my existing wifi on the boat and have the address assigned via a dhcp reservation

mgrouch, 2022-08-11:
To set up wifi without being connected to a network you need monitor and keyboard connected to pi.
So it's not headless at that time. Headless means network and no monitor.

You can script it by putting a script into /boot before first boot.
See: https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_customizing_on_first_boot

You would need to figure out the script though.

martincaddick, 2022-08-11:
gotcha. Burning the image with the raspberry pi imager allows configuration. Just thought that may be suitable.
[image]

mgrouch, 2022-08-11:
I haven't tested that scenario. It's good idea to make it work too. I haven't tried it though so I can't say it's supported at the moment. Might be for future Bullseye based releases.

RichFind, 2022-09-13:
> I haven't tested that scenario. It's good idea to make it work too. I haven't tried it though so I can't say it's supported at the moment. Might be for future Bullseye based releases.

I tried using it, didn't work.

mgrouch, 2022-09-13:
Definitely no plans to support it as it changes user too. 
The BBN documentation now mentions that image needs to be burned as-is without using advanced rpi imager features.

mgrouch, 2022-09-13:
Added proper SD card burning procedure to the documentation. 
I do not see it feasible to test all rpi imager advanced options combinations which modify already tested image.

closed 2022-09-13 by mgrouch