bareboat-necessities/lysmarine_gen #254: Please use the tooling that Signal K Server uses

tkurki, 2022-10-13:
Since you are packaging Signal K server it would be wise to use the tooling that Signal K uses internally. End users turn to Signal K forums for support, and if bareboat is doing things in its own way we waste effort figuring things out.

If there are real, demonstrated issues with the way Signal K server is doing things I am happy to consider changing it.

Doing things differently for minor reasons (like installation speed - i have not heard a single complaint about that, and bareboat being an image does not affect end users at all in initial install) will cause problems over time, at least in support work. 

It is hard enough when people with very little expertise give confusing advice on forums, but if fundamentals (like package manager) vary it makes troubleshooting and helping people just needlessly hard.

tkurki, 2022-10-13:
See also #253

mgrouch, 2022-10-13:
There is no way for me to build BBN OS with SignalK without pnpm on CircleCi. It has limit of 1hr build. With npm it takes hours. 
pnpm is about 5 times faster on arm.

closed 2022-10-24 by mgrouch