bareboat-necessities/lysmarine_gen #304: hotspot issues

zjttoefs, 2023-05-29:
In normal operations I would like the BBN pi (lysmarine ?) to serve it's own standalone WIFI network. To connect to headlessly etc.
But for installations and updates I needed to connect to another Wifi network. 
After having connected to another Wifi network as a client, however, I cannot bring the BBN to act as AP again. Selecting the "lysmarine hotspot" configuration in the network manager results in it trying to connect to a network of that name as a client. 

lysmarine-bbn-bullseye_2023-05-03-raspios-arm64.img.xz

mgrouch, 2023-05-29:
You better off using wired connection to your boat router for internet access and leave pi as WiFi access point. Router serves as firewall too.

zjttoefs, 2023-05-29:
Hmm. Fair enough if that's not the intended use. But the fact that you can easily destroy your AP configuration by using the enabled UI tools without a way back feels wrong. 

Openplotter is actually fairly advanced in that respect and allows both AP and client mode to be configured simultaneously.

mgrouch, 2023-05-29:
Well I just tried and it worked just fine. 

Disconnected from hot spot network. 
it scanned work available 
Connected to available 
It created a new one without deleting hotspot 
Disconnected WiFi client 
Selected connect to hidden network 
Selected old hotspot 
Started old hotspot 

worked just fine.

closed 2023-05-29 by mgrouch