bareboat-necessities/lysmarine_gen #305: No libinput devices loading into X

VibroAxe, 2023-05-31:
I'm sure i'm doing something wrong, but as this is a clean install I've yet to work out what.

First boot of BBN and I see no movement from the touchscreen, plugging in a mouse and keyboard, I also have no input from them.

Connecting over SSH and viewing the Xorg.log (attached) it is full of errors when trying to add the /dev/input devices. If I view the debug output from libinput however it can see the devices and the events from them (all done over ssh and I can't get to the local console!). Only other possible read herring is running libinput debug-events as the user only seems the virtual drivers, running as root shows the full device list. 

No changes made beyond the initial install, and I tried reflashing the card to confirm.

I wasn't expecting the mouse & keyboard to be the difficult bit! Any help is muchly appreciated.

Snippet from Xorg, full log here> https://gist.github.com/VibroAxe/a195063fb1bf68d331621be32ff4d7c6
```
[  1233.467] (II) config/udev: Adding input device (unnamed) (/dev/input/event4)
[  1233.467] (**) (unnamed): Applying InputClass "evdev pointer catchall"
[  1233.467] (**) (unnamed): Applying InputClass "libinput pointer catchall"
[  1233.467] (II) Using input driver 'libinput' for '(unnamed)'
[  1233.525] (II) systemd-logind: got fd for /dev/input/event4 13:68 fd 64 paused 0
[  1233.525] (**) (unnamed): always reports core events
[  1233.525] (**) Option "Device" "/dev/input/event4"
[  1233.525] (**) Option "_source" "server/udev"
[  1233.525] (EE) client bug: Invalid path /dev/input/event4
[  1233.525] (EE) libinput: (unnamed): Failed to create a device for /dev/input/event4
```

libinput debug events
```
user@lysmarine:~ $ libinput debug-events
-event11  DEVICE_ADDED     vc4                               seat0 default group1  cap:kp left scroll-nat scroll-button
-event12  DEVICE_ADDED     vc4                               seat0 default group1  cap:kp left scroll-nat scroll-button
-event13  DEVICE_ADDED     Simulated Right Button            seat0 default group2  cap:p left scroll-nat
^C

user@lysmarine:~ $ sudo libinput debug-events
-event0   DEVICE_ADDED     HID 0e8f:2517                     seat0 default group1  cap:k
-event1   DEVICE_ADDED     HID 0e8f:2517 Consumer Control    seat0 default group1  cap:kp scroll-nat
-event2   DEVICE_ADDED     HID 0e8f:2517                     seat0 default group1  cap:k
-event3   DEVICE_ADDED     HID 0e8f:2517 Mouse               seat0 default group1  cap:p left scroll-nat scroll-button
-event4   DEVICE_ADDED     Razer Razer Mamba Tournament Edition seat0 default group2  cap:p left scroll-nat scroll-button
-event5   DEVICE_ADDED     Razer Razer Mamba Tournament Edition Keyboard seat0 default group2  cap:kp scroll-nat
-event7   DEVICE_ADDED     Razer Razer Mamba Tournament Edition seat0 default group2  cap:k
-event8   DEVICE_ADDED     wch.cn USB2IIC_CTP_CONTROL        seat0 default group3  cap:t  size 528x301mm ntouches 10 calib
-event9   DEVICE_ADDED     wch.cn USB2IIC_CTP_CONTROL Keyboard seat0 default group3  cap:k
-event10  DEVICE_ADDED     wch.cn USB2IIC_CTP_CONTROL        seat0 default group3  cap:p  size 3277x3277mm left scroll-nat calib
-event11  DEVICE_ADDED     vc4                               seat0 default group4  cap:kp left scroll-nat scroll-button
-event12  DEVICE_ADDED     vc4                               seat0 default group4  cap:kp left scroll-nat scroll-button
-event13  DEVICE_ADDED     Simulated Right Button            seat0 default group5  cap:p left scroll-nat
-event0   KEYBOARD_KEY     +0.000s      *** (-1) pressed
 event0   KEYBOARD_KEY     +0.176s      *** (-1) released
 event0   KEYBOARD_KEY     +0.416s      *** (-1) pressed
 event0   KEYBOARD_KEY     +0.488s      *** (-1) released
 event0   KEYBOARD_KEY     +0.696s      *** (-1) pressed
 event0   KEYBOARD_KEY     +0.824s      *** (-1) released
-event8   TOUCH_DOWN       +24.660s     0 (0) 88.02/57.59 (465.19/173.13mm)
 event8   TOUCH_FRAME      +24.660s
 event8   TOUCH_UP         +24.732s
 event8   TOUCH_FRAME      +24.732s
-event4   POINTER_BUTTON   +179.062s    BTN_LEFT (272) pressed, seat count: 1
 event4   POINTER_MOTION   +179.094s      0.00/  0.30 ( +0.00/ +1.00)
 event4   POINTER_MOTION   +179.110s      1.77/  0.88 ( +2.00/ +1.00)
 event4   POINTER_MOTION   +179.126s      2.00/  3.00 ( +2.00/ +3.00)
 event4   POINTER_MOTION   +179.142s      4.00/  3.00 ( +4.00/ +3.00)
 event4   POINTER_MOTION   +179.158s      5.04/  5.04 ( +5.00/ +5.00)
 event4   POINTER_MOTION   +179.302s     -1.00/ -1.00 ( -1.00/ -1.00)
 event4   POINTER_MOTION   +179.318s     -0.99/  0.00 ( -1.00/ +0.00)
 event4   POINTER_BUTTON   +179.318s    BTN_LEFT (272) released, seat count: 0
 event4   POINTER_BUTTON   +179.526s    BTN_LEFT (272) pressed, seat count: 1
 event4   POINTER_MOTION   +179.574s      0.63/  0.00 ( +1.00/ +0.00)
 event4   POINTER_MOTION   +179.590s      0.00/  0.63 ( +0.00/ +1.00)
 event4   POINTER_MOTION   +179.606s      3.95/  1.98 ( +4.00/ +2.00)
 event4   POINTER_MOTION   +179.622s      5.00/  0.00 ( +5.00/ +0.00)
 event4   POINTER_MOTION   +179.638s      6.00/ -1.00 ( +6.00/ -1.00)
 event4   POINTER_MOTION   +179.654s      4.00/ -1.00 ( +4.00/ -1.00)
 event4   POINTER_MOTION   +179.670s      1.00/ -1.00 ( +1.00/ -1.00)
 event4   POINTER_BUTTON   +179.686s    BTN_LEFT (272) released, seat count: 0
 event4   POINTER_MOTION   +179.702s     -0.87/  0.00 ( -1.00/ +0.00)
 event4   DEVICE_REMOVED   Razer Razer Mamba Tournament Edition seat0 default group2  cap:p
-event5   DEVICE_REMOVED   Razer Razer Mamba Tournament Edition Keyboard seat0 default group2  cap:kp
-event7   DEVICE_REMOVED   Razer Razer Mamba Tournament Edition seat0 default group2  cap:k
-event5   DEVICE_ADDED     Razer Razer Mamba Tournament Edition Keyboard seat0 default group6  cap:kp scroll-nat
-event7   DEVICE_ADDED     Razer Razer Mamba Tournament Edition seat0 default group6  cap:k
-event4   DEVICE_ADDED     Razer Razer Mamba Tournament Edition seat0 default group6  cap:p left scroll-nat scroll-button
 event4   POINTER_BUTTON   +224.921s    BTN_LEFT (272) pressed, seat count: 1
 event4   POINTER_MOTION   +224.937s      0.30/  0.00 ( +1.00/ +0.00)
 event4   POINTER_MOTION   +224.953s      0.88/  1.77 ( +1.00/ +2.00)
 event4   POINTER_MOTION   +224.969s      3.01/  6.02 ( +3.00/ +6.00)
 event4   POINTER_MOTION   +224.985s      4.68/ 35.90 ( +3.00/+23.00)
 event4   POINTER_MOTION   +225.001s    -12.00/ 46.00 ( -6.00/+23.00)
```

VibroAxe, 2023-05-31:
Probably should have mentioned>

RPI 4b 8GB

lsusb output
```
user@lysmarine:~ $ sudo lsusb
Bus 002 Device 003: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 002 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 018: ID 1d5c:7102 Fresco Logic Generic Billboard Device
Bus 001 Device 022: ID 32d7:0001 wch.cn USB2IIC_CTP_CONTROL
Bus 001 Device 016: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 023: ID 1532:0046 Razer USA, Ltd Mamba 2015 Tournament Edition [RZ01-01370100-R3]
Bus 001 Device 010: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 008: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 006: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 005: ID 0e8f:2517 GreenAsia Inc.
Bus 001 Device 004: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 003: ID 1547:1000 SG Intec Ltd & Co KG SG-Lock[U2]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

BBN Version
Same issue on `lysmarine-bbn-bullseye_2023-01-19-raspios-arm64.img` and `lysmarine-bbn-bullseye_2023-05-17-raspios-arm64.img`

mgrouch, 2023-05-31:
Looks like hardware issue to me. Unplug all your hubs. Plug just touchscreen, keyboard and mouse directly into pi. And boot. probably better to burn fresh SD card cause you had some usb issues on previous first boot which does some postinstall stuff. If your keyboard, mouse and touchscreen working plugged in directly then start plugging other devices one by one. You might have bad usb cables or hub or power shortage if usb hubs do not supply own power. Sent from my iPhone

mgrouch, 2023-05-31:
Also check kernel messages from dmesgThere shouldn’t be any errors. If there are -  chances are you having issues with hardware. Sent from my iPhone

VibroAxe, 2023-06-01:
I'll try that tonight, this is a fresh SD card. 

Surely the libinput debug wouldn't work if there were underlying hardware issues?

mgrouch, 2023-06-09:
Did it work?
Thanks

mgrouch, 2023-06-15:
Actually, this one could be relevant https://github.com/bareboat-necessities/lysmarine_gen/issues/308

VibroAxe, 2023-06-15:
Apologies it's been a pretty hairy free weeks at work I've but had time to try this. Will get some time asap. Appreciate the early answers. How do I try the above fix, download the latest nightly?

mgrouch, 2023-06-15:
Yes try 2023-06-13 version. 
thanks

VibroAxe, 2023-06-20:
So managed to try the 06-13 version and still had the same issue with no input into the X session (keyboard does work at the console). I tried installing the base Rasbian Bullseye 11 install and that works perfectly so I don't believe this is an OS issue. 

Any thoughts?

mgrouch, 2023-06-20:
How your

/usr/share/X11/xorg.conf.d/90-touchinput.conf

looks like?

Try changing kms to fkms in
your /boot/config.txt

Can you look in logs for any relevant errors?

thanks

mgrouch, 2023-06-20:
On BBN OS. Did you try different keyboard?

any errors in 

dmesg

output?

mgrouch, 2023-06-20:
Can you try plugging 
mouse and keyboard directly into pi usb?

try different usb ports usb2 vs usb3

VibroAxe, 2023-06-20:
i have 2 mice, a keyboard and the touchscreen. None appear to give anything into X, however console can see all of them as root libinput debug. Let me reburn the BBNOS image and grab some more log files over ssh

VibroAxe, 2023-06-20:
hdmi, you on discord somewhere to drop this off github as chat :D

mgrouch, 2023-06-20:
there is 2023-06-15 now. 
In it I switched back to default fkms driver instead of kms.
kms was having issues with some DSI connected screens. You might try that one since your are burning new one.

VibroAxe, 2023-06-20:
> there is 2023-06-15 now.

Downloading now

mgrouch, 2023-06-20:
Any errors in

dmesg

?

VibroAxe, 2023-06-20:
Don't remember any, 85% written on new image so will do a full log collection again in a sec and upload

VibroAxe, 2023-06-20:
Right, @mgrouch 
New log files uploaded to https://gist.github.com/VibroAxe/4ee9251922dfb5a48fc39b0768b30cf6

BBNOS 2023-06-15

Again, `libinput debug-events` is showing no input devices (and no events), where as `sudo libinput debug-events` is showing all the devices and events expected

Devices are connected via usb3 in the attached files. i've just tried switching to usb2 and no difference

VibroAxe, 2023-06-20:
NB: fkms is enabled already in 06-15 (I believe expected)
```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d,cma-128
max_framebuffers=2
```

mgrouch, 2023-06-20:
There are errors in Xorg log

somehow it say can’t use /dev/input/event*
files

mgrouch, 2023-06-20:
libinput: (unnamed): Failed to create a device for /dev/input/event10

And so on

VibroAxe, 2023-06-20:
Further diagnostics in a similar vein: if I stop the startx service, and then run /usr/bin/startx manually as root, all input works, however I obviously get a root X session without gnome or any bbnos elements

Doing the same as user, gives me the BBNOS shell, but no input devices!

VibroAxe, 2023-06-20:
```
user@lysmarine:~ $ ls -al /dev/input/event*
crw-rw---- 1 root input 13, 64 Jun 20 20:19 /dev/input/event0
crw-rw---- 1 root input 13, 65 Jun 20 20:19 /dev/input/event1
crw-rw---- 1 root input 13, 74 Jun 20 20:32 /dev/input/event10
crw-rw---- 1 root input 13, 66 Jun 20 20:19 /dev/input/event2
crw-rw---- 1 root input 13, 67 Jun 20 20:19 /dev/input/event3
crw-rw---- 1 root input 13, 68 Jun 15 20:49 /dev/input/event4
crw-rw---- 1 root input 13, 69 Jun 15 20:49 /dev/input/event5
crw-rw---- 1 root input 13, 70 Jun 15 20:49 /dev/input/event6
crw-rw---- 1 root input 13, 71 Jun 15 20:49 /dev/input/event7
crw-rw-r-- 1 root input 13, 72 Jun 20 20:32 /dev/input/event8
crw-rw---- 1 root input 13, 73 Jun 20 20:32 /dev/input/event9
```

```
user@lysmarine:~ $ grep input /etc/group
input:x:104:user,lirc
```

Seems to imply the permissions should be present

mgrouch, 2023-06-20:
Is user ‘user’ member of group ‘input’?

type 

groups

VibroAxe, 2023-06-20:
```
user@lysmarine:~ $ groups
user adm tty lp dialout fax voice cdrom sudo audio video plugdev games users input netdev bluetooth lpadmin geoclue adbusers lirc gpio i2c spi pypilot charts raspotify solarthing
```

yup

mgrouch, 2023-06-20:
I’m checking how it’s looking on my installation where it works

mgrouch, 2023-06-20:
I do not have that many files in /dev/input

only events 0-3

mgrouch, 2023-06-20:
I do not have those errors marked EE
in Xorg log as you have.

VibroAxe, 2023-06-20:
wch.cn is my touchscreen, Razer Mamba is the mouse and HID 0e8f is the random touchpad/keyboard combo thing i've got plugged in

VibroAxe, 2023-06-20:
OOI, on your working system, what happens if you run `libinput debug-events` from either console/ssh as `user`? Mine can only see the simulated right button

mgrouch, 2023-06-20:
How your

/usr/share/X11/xorg.conf.d/90-touchinput.conf

looks like?

VibroAxe, 2023-06-20:
```
user@lysmarine:~ $ cat /usr/share/X11/xorg.conf.d/90-touchinput.conf
Section "InputClass"
    Identifier "calibration"
    Driver "evdev"
    MatchProduct "wch.cn USB2IIC_CTP_CONTROL"
    MatchDevicePath "/dev/input/event*"
    Option "EmulateThirdButton" "1"
    Option "EmulateThirdButtonTimeout" "750"
    Option "EmulateThirdButtonMoveThreshold" "30"
EndSection
```

mgrouch, 2023-06-20:
try

killall evdev-rce
killall twofing

and after that try keyboard and mouse

mgrouch, 2023-06-20:
Sounds like this 

https://github.com/systemd/systemd/issues/14027

VibroAxe, 2023-06-20:
```
user@lysmarine:~ $ killall evdev-rce
user@lysmarine:~ $ killall twofing
twofing: no process found
```

No dice. However, some progress!
If I remove the /usr/share/X11/xorg.conf.d/40-libinput.conf  file (which declares the input/event devices as libinput) and reboot X, I now get touchscreen and keyboard/mouse and your touch and hold to right click. This looks to be some oddity of legacy drivers for libinput conflicting with the evdev drivers in 10-evdev.conf

Minor issue is that I seem to need to hotplug the keyboard/mouse to get that to work, but theres some progress here. I'm thinking you have some deconfliction to do in the default device map somewhere?

mgrouch, 2023-06-20:
Any other errors in journal

journalctl -p err -b

VibroAxe, 2023-06-20:
Several. But nothing to do with X and I suspect all to do with requiring configuration futher down the line

```
-- Journal begins at Thu 2023-06-15 20:49:50 UTC, ends at Tue 2023-06-20 21:35:57 UTC. --
Jun 15 20:49:50 lysmarine systemd-modules-load[159]: Failed to find module 'lp'
Jun 15 20:49:50 lysmarine systemd-modules-load[159]: Failed to find module 'ppdev'
Jun 15 20:49:50 lysmarine systemd-modules-load[159]: Failed to find module 'parport_pc'
Jun 15 20:49:51 lysmarine systemd[1]: Failed to start Uncomplicated firewall.
Jun 15 20:49:58 lysmarine systemd-udevd[261]: ttyUSB0: Failed to create symlink '/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2.2.2:1.0-port0.tmp-c188:0' to '../../ttyUSB0': Not a directory
Jun 15 20:50:00 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 15 20:50:00 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 15 20:50:00 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 15 20:50:00 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 15 20:50:05 lysmarine pipewire[1497]: Failed to receive portal pid: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get PID of name 'org.freedesktop.portal.Desktop': no such name
Jun 15 20:50:05 lysmarine bluetoothd[1539]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Jun 15 20:50:05 lysmarine bluetoothd[1539]: sap-server: Operation not permitted (1)
Jun 15 20:50:05 lysmarine bluetoothd[1539]: Failed to set privacy: Rejected (0x0b)
Jun 15 20:50:05 lysmarine pulseaudio[1499]: Source alsa_output.platform-bcm2835_audio.analog-mono.2.monitor does not exist.
Jun 20 21:14:30 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:14:30 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:14:40 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:14:40 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:14:42 lysmarine wpa_supplicant[1224]: Using interface wlan0 with hwaddr e4:5f:01:3e:16:95 and ssid "lysmarine-hotspot"
Jun 20 21:14:50 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:14:50 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:15:00 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:15:00 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:15:10 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:15:10 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:15:20 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:15:20 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:15:27 lysmarine pulseaudio[2811]: Source alsa_output.platform-bcm2835_audio.analog-mono.2.monitor does not exist.
Jun 20 21:15:30 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:15:30 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:15:40 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:15:40 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:15:50 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:15:50 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:16:00 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:16:00 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:16:10 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:16:10 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:16:20 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:16:20 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:16:30 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:16:30 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:16:40 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:16:40 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 20 21:16:50 lysmarine motion[792]: [1:ml1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/video0: No such file or directory
Jun 20 21:16:50 lysmarine motion[792]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
```

mgrouch, 2023-06-20:
How many devices you have plugged in?
Does it work if you boot only with 
Keyboard mouse and screen?

VibroAxe, 2023-06-20:
I've only got a keyboard, mouse and screen connected. Same issue if I unplug the 2nd mouse. No other devices are connected at the moment

mgrouch, 2023-06-20:
Do you have another keyboard to try?
I wonder why it works for others including me

VibroAxe, 2023-06-20:
What HW revision of Pi you running, a 4?

Will see what I can find lying around. I did try with just the touchscreen connected but no success there either

mgrouch, 2023-06-20:
Yes mine is pi4 8Gb

mgrouch, 2023-06-20:
Revision 1.4

VibroAxe, 2023-06-20:
Unrelated question whilst I continue to debug. Is there a way to configure BBN to start the gui without the display connected. I'll always have the same display so the resolution can be hardcoded. I don't have the screen out all the time on my boat, but often will VNC from a tablet or use the various APIs (i'm migrating from an openplotter system).

The only reference to headless in the manual I could find was talking about  not running the UI at all, which isn't what I had in mind

mgrouch, 2023-06-20:
You need to buy hdmi fake plug.

mgrouch, 2023-06-20:
Hdmi dummy plug. They are about $3

mgrouch, 2023-06-20:
Or actually better 

set 
`
hdmi_force_hotplug=1
`
in /boot/config.txt

mgrouch, 2023-06-21:
I've tried another pi. Older revision Pi 4 B rev 1.1
And I see same issue.

Deleting  /usr/share/X11/xorg.conf.d/40-libinput.conf makes mouse work in X session 
(but not keyboard) Keyboard starts working after hot re-plug

mgrouch, 2023-06-21:
sudo apt install udev/bullseye-backports

seems fixed it for me. It installed 252.5-2~bpo11+1

It seems 247.3-7+deb11u2 is broken

mgrouch, 2023-06-21:
Can you try image 2023-06-21? I think it should fix the issue

VibroAxe, 2023-06-22:
Will try, but I need another SD card as this one now has "content" on it :P.  SD card has been ordered but wont be here till Saturday and I'm on holiday till Tuesday but will try asap. 

Thanks for the continued hard work!

mgrouch, 2023-06-22:
You can also try restoring the Xorg configuration file 40-something 
you moved out and 

run
`
sudo apt install udev/bullseye-backports
`

followed by reboot with your current SD card.

closed 2023-07-07 by mgrouch