bareboat-necessities/lysmarine_gen #307: Failed to start Telegraf

ryansabin, 2023-06-08:
I used the rpi-clone utility to start using an ssd. After this change I am getting "failed to start Telegraf" 4 times before startx is running.

user@lysmarine:~ $ sudo systemctl status telegraf.service
● telegraf.service - Telegraf
     Loaded: loaded (/lib/systemd/system/telegraf.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/telegraf.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Wed 2023-06-07 23:57:23 MDT; 5min ago
       Docs: https://github.com/influxdata/telegraf
   Main PID: 1394 (code=exited, status=1/FAILURE)
        CPU: 423ms

Jun 07 23:57:23 lysmarine systemd[1]: telegraf.service: Scheduled restart job, restart counter>
Jun 07 23:57:23 lysmarine systemd[1]: Stopped Telegraf.
Jun 07 23:57:23 lysmarine systemd[1]: telegraf.service: Start request repeated too quickly.
Jun 07 23:57:23 lysmarine systemd[1]: telegraf.service: Failed with result 'exit-code'.
Jun 07 23:57:23 lysmarine systemd[1]: Failed to start Telegraf.
Jun 08 00:02:57 lysmarine systemd[1]: /etc/systemd/system/telegraf.service.d/override.conf:1: >
Jun 08 00:02:57 lysmarine systemd[1]: /etc/systemd/system/telegraf.service.d/override.conf:2: >
Jun 08 00:03:16 lysmarine systemd[1]: /etc/systemd/system/telegraf.service.d/override.conf:1: >
Jun 08 00:03:16 lysmarine systemd[1]: /etc/systemd/system/telegraf.service.d/override.conf:2: >

mgrouch, 2023-06-08:
Looks like you forgot to delete old partition and you cloned to SSD over some old data existing on it. 

Did you follow “Getting Started Guide”?

`SSD Boot  | Before doing it (if you had a previous installation), go into Disks tool from App Menu/Utilities/Disks and remove all partitions from your SSD drive.`

ryansabin, 2023-06-08:
I have tried again using fdisk to delete partitions then running rpi-clone. Still the same issue of tlegraf not starting.

root@lysmarine:/home/user# fdisk /dev/sda

Welcome to fdisk (util-linux 2.36.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): d
Partition number (1,2, default 2): 2

Partition 2 has been deleted.

Command (m for help): d
Selected partition 1
Partition 1 has been deleted.

Command (m for help): p
Disk /dev/sda: 931.51 GiB, 1000204886016 bytes, 1953525168 sectors
Disk model: 2115            
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 33553920 bytes
Disklabel type: dos
Disk identifier: 0xc198e278

Command (m for help): w
The partition table has been altered.

Calling ioctl() to re-read partition table.
Syncing disks.

ryansabin, 2023-06-08:
I cloned a fresh image of the 5-17 build bbn-os. This worked without issue. I am not sure if I modified something in the last build or it was something in the build.

Thanks for responding. I just started working with your build and am very excited to use it more. Great job.

I plan to put a build document for my build shortly.

mgrouch, 2023-06-08:
Did you load any other software yourself after booting from BBN SD card?

telegraf is disabled on it. It wouldn’t attempt to start unless you have done something to enable it. 

also file paths for telegraf in your log 
do not match what is on BBN OS. 

I’ve tried starting telegraf on mine it starts fine. 
but its script is in

/etc/init.d/telegraf

so somehow you have configuration of your telegraf which didn’t come from BBN OS image, but from some other system. 

You could have better idea how that could happen because you know what other steps you did. 

I’d not trust this configuration. I would burn new SD card, erased all partitions from SSD and start loading system from scratch.

ryansabin, 2023-06-08:
I am starting from scratch with a clean burn. Here is the commands I ran on the other version. I use vim for editing and barrier to share the mouse and keyboard from my laptop. I don't see why vim or barrier would try to start telegraf but I will look into that.

~/add-ons/change-password.sh 
sudo passwd root
~/add-ons/hot-fixes-install.sh 
~/add-ons/timezone-setup.sh 
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install vim -y
sudo apt-get install barrier -y
cp /usr/share/applications/barrier.desktop ~/.config/autostart/
vim ~/.config/autostart/barrier.desktop 
mkdir .barrier
vim ~/.barrier/run-barrier.sh
chmod +x ~/.barrier/run-barrier.sh 
sudo rpi-clone sda

mgrouch, 2023-06-08:
This is what I do before 

rpi-clone

SSD Boot  | Before doing it (if you had a previous installation), go into Disks tool from App Menu/Utilities/Disks and remove all partitions from your SSD drive.

mgrouch, 2023-06-08:
apt upgrade 

is also unnecessary. It does break things sometimes. 

If something doesn’t work for you and and there is known fix via apt upgrade
I would just cherry pick package that needs to be upgraded and install only that one and dependencies via apt.

ryansabin, 2023-06-08:
Thanks for the heads up on that upgrade. I can upgrade individually as/if needed.

mgrouch, 2023-06-09:
You upgraded telegraf

doing 
apt upgrade 

somehow that upgrade breaks it. 
:)

not all upgrades are safe in fact.

closed 2023-06-26 by mgrouch