bareboat-necessities/lysmarine_gen #313: Boat app locked up

Tosca36, 2023-06-15:
I entered all the boat information again as it didn't save it and tried to go back in and I can't open it, I can open other apps like signalk, radio and others, I restarted the pi and still not able to open the boat app? Any help would be much appreciated!

mgrouch, 2023-06-15:
Do

rm -r /home/user/.vessel 

and start app. 
enter your data without using special characters 

like quotes etc

Tosca36, 2023-06-17:
Hello Mikhail! I ran that command just like you have it written and went back into vessel app and still couldn’t open it. Any suggestions? Thank you john

Tosca36, 2023-06-17:
Sorry! Also opencpn is running very slow, is there a way to speed it up? I can’t find the settings for OpenGL

Tosca36, 2023-06-17:
Hello Mikhail! What should I choose to make opencpn faster,also did grip command in the terminal and open cpn shows an error?  also it doesn’t seem to scale the charts when you pan closer? Thank you Mikhail!!! JohnThank you john

mgrouch, 2023-06-17:
You need to have BBN OS

version 2023-06-15  (or 13)

for OpenCPN to run fast.

Tosca36, 2023-06-17:
Do I need to start over with the new version or can I update what I have now?

mgrouch, 2023-06-18:
You need to install new image and apply same changes you done before. Hopefully you scripted them somewhere or just made notes how to reproduce them.

closed 2023-12-19 by mgrouch