bareboat-necessities/lysmarine_gen #317: i lost my own position and own vessel colored by grey

Topaz12244, 2023-06-26:
hi, we just lost all ais, own vessel position and now became grey and position cannot detect.

we have checked all connections are good from AIS port to PC any suggestions.?

mgrouch, 2023-06-26:
Try restarting SignalK. 
click command icon on desktop and there is option to restart SignalK from touchscreen

mgrouch, 2023-06-26:
If that doesn’t help. Reboot pi

Topaz12244, 2023-06-26:
> 

i opened command but there is no option to restart singalK

mgrouch, 2023-06-26:
Click an icon on desktop tilted ‘Command’

Topaz12244, 2023-06-26:
> Click an icon on desktop tilted ‘Commands’ Third row. Middle

hi dear, i can't find any icon on my pc desktop named by command.

mgrouch, 2023-06-26:
[image]

mgrouch, 2023-06-26:
Third row middle

Topaz12244, 2023-06-26:
> [image]

thank you for trying help me .

moreover i`m installed opencpn on computer PC in my boat and connected AIS to CPU of the computer by Cable.

closed 2023-06-29 by mgrouch