bareboat-necessities/lysmarine_gen #331: SignalK - Updating or removing Apps

wsturner88, 2023-07-23:
Does BBN limit the ability to remove applications from SignalK?      I have a minimal understanding of Signalk but wanted to remove a number of Apps while I try to integrate this with my NMEA 2000 Network.    

Also was going to upgrade to 2.1 but if fails each time I try.

mgrouch, 2023-07-23:
For SignalK upgrades you are better off
loading new image. The latest image has SignalK 2.1. 

Updates/delete of plugins should work from SignalK. 

I checked upgrading of one plugin and it worked for me. 

Delete somehow didn’t work. It looks like there is a bug in deletes. 

Thanks

mgrouch, 2023-07-23:
BTW you can see how SignalK is installed from this script 

https://github.com/bareboat-necessities/lysmarine_gen/blob/bullseye/install-scripts/4-server/04-signalk.sh

wsturner88, 2023-07-23:
If I update, is there a way to save and restore all of my settings?

Especially Kip and open on

mgrouch, 2023-07-23:
There is SignalK back up / restore functionality. 

When you click on ‘backup’ it creates backup file in ~/Downloads directory.

mgrouch, 2023-07-23:
You can run:

````
cd /home/user/add-ons
./hot-fixes-install.sh
````

on existing installation to fix 'deleting' plugins issue

closed 2023-07-23 by mgrouch