bareboat-necessities/lysmarine_gen #335: onboard - issues displaying the on-screen keyboard

christianstrauch, 2023-08-12:
I am experiencing display issues using the **onboard** on-screen keyboard as shown in the attached images. Modifying text/key sizes does not resolve the issue. Neither does removing and reinstalling the onboard-common, onboard, onboard-data packages.
[image]
[image]

mgrouch, 2023-08-12:
What is the output of

`
cat   /etc/bbn-build
`

mgrouch, 2023-08-12:
Does this problem happen when you connect monitor directly without VNC?

christianstrauch, 2023-08-12:
- /etc/bbn-build contains 2023-07-23 18:29:49+00:00
- The problem occurs with or without VNC
- This is a same-day, fresh installation

mgrouch, 2023-08-12:
Can you post files you have in
`
/home/user/.config/openbox/
`
?

christianstrauch, 2023-08-12:
My monitor is a 9" 1080p touchscreen (chalkelec) connected via HDMI and USB.
I did modify a few things, but nothing fundamental. I'm intending to change the template for the panel (including different icons) - that's why I removed the credit (with the night controls which I'll bring back soon). Everything else in the UI (except for onboard) is working, however:

The screen hadn't been detected on first boot, but it was on a subsequent reboot. It was then used with 720p resolution. Switching to 1080p works, too.
xrandr says:
`Screen 0: minimum 320 x 200, current 1280 x 720, maximum 7680 x 7680
HDMI-1 connected primary 1280x720+0+0 (normal left inverted right x axis y axis) 16mm x 9mm
   1280x720      60.00*+  50.00    59.94
   1920x1080     60.00    50.00    59.94    30.00    29.97
   1920x1080i    60.00    50.00    59.94
   1024x768      60.00
   800x600       60.32
   720x576       50.00
   720x480       60.00    59.94
   640x480       60.00    59.94
`
Executing `xrandr --output HDMI-1 --mode 1920x1080` resizes the screen perfectly fine but onboard still behaves weird.  
- The touchscreen had been recognized as "tablet" by X which obviously wouldn't work so I had to create an udev rule to fix that. It now works and is calibrated

mgrouch, 2023-08-12:
Can you exit onboard and just run it from command line in terminal window to see if it prints any errors?

christianstrauch, 2023-08-12:
It showed up fine at first but broke while using it. Here are the warnings generated when running it interactively.
[image]
At this point, I'm not sure I'm still using the originally selected theme tho - I switched themes to see if there was an issue with any of them (but that didn't help).

mgrouch, 2023-08-12:
Looks like result of you playing with keyboard language settings. You seem replaced ibus (default language manager) with something else. Per screenshots all letters disappeared on your keyboard. 

I suggest full reinstall and using ibus for adding additional language keyboards.

christianstrauch, 2023-08-12:
Where are you getting this assumption about keyboard language settings from? I have not "played" with keyboard language whatsoever (English (US) has been the only one installed and never changed).
I am a bit worried that this would happen randomly during operations - since it also appeared without me changing anything before. I am not sure re-imaging the SD card would solve this long term?
What makes you think reinstalling would help? Which installation step do you assume needs to be repeated?

christianstrauch, 2023-08-12:
Also, the letters did not disappear on the keyboard - the screenshot you are referring to shows the numeric keyboard. The issue mostly is around sizing - the size of the keyboard keeps changing and seems to be initially set to a very large size (something like 10 times the screen size). Also, if (sometimes) the keyboard starts vaguely appropriately, it shows a much larger version of itself in the background (see first screenshot).

christianstrauch, 2023-08-13:
Issue is (temporarily, and only for me) fixed:
My HDMI touchscreen device registers its physical size as 16x9 which onboard interprets as 16mm x 9mm. I patched onboard's WindowUtils.py to account for that:
```python
def get_monitor_dimensions(window):
    """ Geometry and physical size of the monitor at window. """
    gdk_win = window.get_window()
    screen = window.get_screen()
    if gdk_win and screen:
        monitor = screen.get_monitor_at_window(gdk_win)
        r = screen.get_monitor_geometry(monitor)
        size = (r.width, r.height)
        size_mm = (screen.get_monitor_width_mm(monitor),
                   screen.get_monitor_height_mm(monitor))

        # Nexus7 simulation
        device = None       # keep this at None
        # device = 1

        # Chalkelec hack
        # There was already a "Nexus7 simulation" in here which coincidentally requires the same solution
        device = 2
        if device == 0:     # dimension unavailable
            size_mm = 0, 0
        elif device == 1:     # Nexus 7, as it should report
            size = 1280, 800
            size_mm = 150, 94
        elif device == 2:   # Chalkelec hack - the screen reports 16, 9 for some reason
            size_mm = 150, 94
        return size, size_mm
    else:
        return (0, 0), (0, 0)
```

mgrouch, 2023-08-13:
On my installation I do not see the errors you are seeing when I start onboard from command line. 

The first one is related to keyboard language. 

But now I realized it might be due to you using VNC and might be unrelated.

mgrouch, 2023-08-13:
Is your keyboard layout English US on the computer you VNC from?

mgrouch, 2023-08-13:
Are you setting

hdmi_force_hotplug=1

in /boot/config.txt
 
?

mgrouch, 2023-08-13:
Might be when you are booting with hdmi screen off and then VNC it just doesn’t know what are your screen parameters

christianstrauch, 2023-08-13:
Looking at config.txt now.
Regarding the other questions:
- The HDMI device is constantly plugged in. It is generally detected quite early, too.
- Keyboard layouts are English (US) everywhere - on the Pi, on the clients I use with VNC.
- VNC may or may not be connected when X starts - often it wasn't over the previous days. Doesn't change anything.

I'll update this thread once I'm done going through /boot/config.txt.

mgrouch, 2023-08-19:
You can change headless resolution for VNC using raspi-config tool. 

See: https://www.radishlogic.com/raspberry-pi/how-to-set-the-default-screen-resolution-for-vnc-viewer-when-raspberry-pi-is-not-connected-to-a-monitor/#Terminal

closed 2023-08-24 by mgrouch