bareboat-necessities/lysmarine_gen #341: Open cpn error

Tosca36, 2023-08-31:
Hello everyone! Started again with the new BBN os 
8/17/2023, downloaded from cloudsmith. I thought I was having issues with signalk. After boot up, I went to signalk to check open cpn and have a connect ECONNREFUSED 127.0.0.1:30330 error. I had this also on my last BBN os download, am I doing something wrong here? Do I need to turn an app on or configure my connections first?? I also cant download all the info for overlays etc. for the Marine Control Server I am using, I never had that issue before.do I need to go back to an older version of BBN os to have it work? Thank you!!!

mgrouch, 2023-08-31:
You should ignore this error. It’s normal. 

Port 30330 is open only when OpenCPN is running. This is how OpenCPN sends waypoint information when route is activated in OpenCPN.

mgrouch, 2023-08-31:
For your MCS related issues you should report them at  https://github.com/Thomas-GeDaD/SignalK_raspberry_MCS

closed 2023-09-01 by mgrouch

Tosca36, 2023-09-01:
Hello Mikhail! In the bareboat necessities os, it says that the app for the MCS board is installed? Is this correct? Then all I need to do is run the post install script to get it to work with signalk? Thank you!!!

mgrouch, 2023-09-01:
Yes

Tosca36, 2023-09-01:
Thank you Mikhail!!