bareboat-necessities/lysmarine_gen #351: SignalK can't install, upgrade, or delete plugins

toastygm, 2023-10-10:
I am unable to install, upgrade, or delete any plugins.  It always fails.  The error generally is:

```
50 timing command:install Completed in 24830ms
51 verbose stack TypeError: Cannot read properties of null (reading 'matches')
51 verbose stack     at Link.matches (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/node.js:1115:41)
51 verbose stack     at Link.canDedupe (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/node.js:1069:15)
51 verbose stack     at PlaceDep.pruneDedupable (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/place-dep.js:426:14)
51 verbose stack     at new PlaceDep (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/place-dep.js:278:14)
51 verbose stack     at #buildDepStep (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/arborist/build-ideal-tree.js:933:18)
51 verbose stack     at async Arborist.buildIdealTree (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/arborist/build-ideal-tree.js:196:7)
51 verbose stack     at async Promise.all (index 1)
51 verbose stack     at async Arborist.reify (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/arborist/reify.js:159:5)
51 verbose stack     at async Install.exec (/usr/lib/node_modules/npm/lib/commands/install.js:151:5)
51 verbose stack     at async module.exports (/usr/lib/node_modules/npm/lib/cli-entry.js:61:5)
52 verbose cwd /home/signalk/.signalk
53 verbose Linux 6.1.21-v8+
54 verbose node v18.17.1
55 verbose npm  v10.1.0
56 error Cannot read properties of null (reading 'matches')
57 verbose exit 1
```
Full log file is [here](https://github.com/bareboat-necessities/lysmarine_gen/files/12858618/errlog.txt).

mgrouch, 2023-10-10:
I've just tried on 2023-08-17 version and it worked for me.
Here is a screenshot
[image]

toastygm, 2023-10-10:
Hmmm.  Well, there are two differences that I can see:

1. I am using BBN OS 2023-07-23 18:29:49+00:00
2. I am running this from HTTPS from another machine, not directly on the BBN machine (this shouldn't really matter)

I'll keep fiddling.

mgrouch, 2023-10-10:
There were few images around your version where SignalK plugin updates/installs were broken in UI on BBN OS.
2023-07-23 might be the one that has it broken. 
2023-08-17 seems ok.

toastygm, 2023-10-10:
Running these commands fixed my problem:

```
# use pnpm instead of npm
sed -i 's#('"'npm',#\('pnpm'"',#' /usr/lib/node_modules/signalk-server/lib/modules.js

# SignalK fix for pnpm
sed -i -e s/--save"'",/"--save-prod'",/g /usr/lib/node_modules/signalk-server/lib/modules.js
```

closed 2023-10-10 by toastygm