bareboat-necessities/lysmarine_gen #353: Install - Pi 4 8GB - 10.1 inch Screen

wsturner88, 2023-10-10:
Having installed BBN about 10 times now, I thought I knew what I was doing.   Looking to upgrade my boat and add a 10" touch screen.   

On first boot, it resizes the partition.. I get the LYSMARINE splash screen..  Then I get a repeating authentication error.
Tried a few builds 64bit all had the same issue. 

Any recommendations?

wsturner88, 2023-10-10:
Good Thought, I have been using a 16GB Samsung USB stick.    For some other applications I found it much faster than the SD card.

YES I did change name and password in imager...      Will try without that as well.

mgrouch, 2023-10-10:
If you changed by imager you will get the error you are receiving

mgrouch, 2023-10-10:
[image]

closed 2023-10-11 by mgrouch

wsturner88, 2023-10-11:
SOLVED