bareboat-necessities/lysmarine_gen #358: Signal K server white screen (not working)

Aitonos, 2023-11-18:
Hi to everybody. 

,Yesterday I couldn't reach kip from other devices in the network of the BBN. 

Trying to figure out how to get it accessible, I checked the ssl connection under signal K settings and after restart, no more access to signal K server, and no more kip, subsequently.

How can I get signal K back to life without losing all the settings?

Thanks!

mgrouch, 2023-11-19:
You can ask on SignalK support slack channel how to undo your changes you did with SSL that broke your system.
Here is the link: https://signalk-dev.slack.com/

Aitonos, 2023-11-19:
Thanks for the reply. In OP there's sk installer that allows y to deactivate ssl, is not any similar tool in BBN? In case I have to reinstall sk, how to be done in BBN?
Thanks for the great work.

Aitonos, 2023-11-20:
Thanks, I'll try.

closed 2023-12-19 by mgrouch