bareboat-necessities/lysmarine_gen #360: Support for Raspberry Pi 5?

tingles2233, 2023-12-12:
As the title says support for Raspberry Pi 5? 

Most likely means a port over to Bookworm from Bullseye.

mgrouch, 2023-12-12:
Let’s see how many people actually need it. 
Please comment 
+1 if you need pi5 version

Starcrest2020, 2023-12-13:
÷1

tingles2233, 2023-12-13:
The big win for me with this is the faster processor which allows for in our case as it is positioned as a multipurpose device we can watch proper full resolution 4K web streams. All other functions would also benefit from the overall speed increase.

mgrouch, 2023-12-13:
> The big win for me with this is the faster processor which allows for in our case as it is positioned as a multipurpose device we can watch proper full resolution 4K web streams. All other functions would also benefit from the overall speed increase.

Speed increase in pi5 comes with higher power consumption as a con.

tingles2233, 2023-12-13:
Not wrong rpi5 idle power consumption is around 3.4watts compared to the idle of the rpi4 at 2.8watts in my testing. 

Other then actively watching a video the pi seems to stay at idle power consumption. Even with opencpn running it doesn’t seem to consume more(on the rpi4) that is I would assume the rpi5 would be the same in that regard.

mgrouch, 2023-12-16:
You can monitor the progress on test images which are appearing here

https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/?q=bookworm

mgrouch, 2023-12-21:
BBN Lite version based on bookworm looks pretty good already. Here are known issues 

- PyPilot calibration doesn’t start
- application Menu organization changed

Report here if you see other issues with BBN Lite

BBN Full version on bookworm is still missing features

tingles2233, 2023-12-28:
Thanks, I’ll be doing testing first week in Jan and report back with my findings.

mgrouch, 2024-01-13:
BBN Lite version based on bookworm looks pretty good already. Here are known issues

- application Menu organization changed
- pypilot motor controller (untested on pi5, it seems to be working on pi4)
- no password popup on connecting to new Wi-Fi

On bookworm version for pypilot motor controller UART you also need to disable the hciuart service, which initialises the Bluetooth modem:

````
sudo systemctl disable hciuart
````

Report here if you see other issues with BBN Lite

BBN Full version on bookworm is still missing features:

- Pypilot calibration tool issues with compass calibration screen
- Mopidy no audio
- no password popup on connecting to new Wi-Fi

mgrouch, 2024-01-13:
This https://github.com/mopidy/mopidy/issues/1974
seems related to mopidy audio issues

mgrouch, 2024-01-16:
Remaining issues

Pypilot calibration tool issues with compass calibration screen (in FULL version)
https://github.com/pypilot/pypilot/issues/216

No password popup on connecting to new Wi-Fi
https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/1461
(as workaround you can enter password later in WiFi settings)

Web cam app doesn't start (seg faults)
https://gitlab.gnome.org/GNOME/cheese/-/issues/174

Application menu organization changed
https://github.com/BuddiesOfBudgie/budgie-desktop/issues/507

mgrouch, 2024-01-29:
No password popup on connecting to new Wi-Fi was fixed with using lightdm

mgrouch, 2024-01-29:
Remaining (known) issues

Pypilot calibration tool issues with compass calibration screen (in FULL version)
https://github.com/pypilot/pypilot/issues/216

Web cam app doesn't start (seg faults) (affects FULL version)
https://gitlab.gnome.org/GNOME/cheese/-/issues/174

Application menu organization changed
https://github.com/BuddiesOfBudgie/budgie-desktop/issues/507
https://github.com/BuddiesOfBudgie/budgie-desktop/issues/506

mgrouch, 2024-02-07:
webcam and app menu organization issues and fixed

mgrouch, 2024-02-07:
Remaining (known) issues

Pypilot calibration tool issues with compass calibration screen
https://github.com/pypilot/pypilot/issues/216

mgrouch, 2024-02-10:
Remaining (known) issues

Pypilot calibration tool issues with compass calibration screen
https://github.com/pypilot/pypilot/issues/216

pigpiod and any pigpio specific features wouldn’t work
on pi5. Seatalk1 plugin which requires pigpiod and pypilot hat (motor controller and IMU should be ok)

mgrouch, 2024-02-11:
The first release with bookworm and support for raspberry pi 5 is out.
Closing

closed 2024-02-11 by mgrouch

mgrouch, 2024-02-12:
pi5 is supported by both LITE and FULL version since Release:

https://github.com/bareboat-necessities/lysmarine_gen/releases/tag/v2024-02-12.1