bareboat-necessities/lysmarine_gen #363: GPS Connection error - New user, may be my issue

Techstyleuk, 2023-12-24:
GPS shows ups as ttyLYS_gps_0 and getting no GPS feed in Signal K.

ls -l /dev/ttyLYS*
lrwxrwxrwx 1 root root 7 Dec 23 19:29 /dev/ttyLYS_gps_0 -> ttyUSB0
lrwxrwxrwx 1 root root 7 Dec 23 19:29 /dev/ttyLYS_prolific_ -> ttyUSB0

updated the GPS1 entry in Signal K from '/dev/ttyLYS_gps_1' to '/dev/ttyLYS_gps_0', then GPS signal successfully received by Signal K.

I am new to BBN so maybe there is a way to handle this similar to Openplotter Serial, so let me know if I am just doing this wrong

mgrouch, 2023-12-24:
Just edit /etc/default/gpsd.conf 

and add 
/dev/ttyLYS_gps_0

to devices

mgrouch, 2024-01-06:
````
DEVICES="/dev/ttyLYS_gps_0"
````

in  /etc/default/gpsd