bareboat-necessities/lysmarine_gen #375: FULL version pypilot calibration issues

mgrouch, 2024-02-11:
Pypilot calibration tool issues with compass calibration screen in FULL version

https://github.com/pypilot/pypilot/issues/216

Calibration (compass and accelerometer) in FULL version is missing spheres in UI.
No calibration data is displayed in left bottom text area.

mgrouch, 2024-02-11:
https://github.com/pypilot/pypilot/issues/220

seandepagnier, 2024-02-11:
Can you check the log file for pypilot?  Is it full of failed send ex??

mgrouch, 2024-02-12:
````
user@lysmarine:~ $ systemctl status pypilot@pypilot
● pypilot@pypilot.service - pypilot
     Loaded: loaded (/etc/systemd/system/pypilot@.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-02-11 22:25:38 UTC; 1h 41min ago
   Main PID: 743 (pypilot)
      Tasks: 14 (limit: 9257)
        CPU: 1min 20.880s
     CGroup: /system.slice/system-pypilot.slice/pypilot@pypilot.service
             ├─ 743 /usr/bin/python3 /usr/local/bin/pypilot
             ├─1617 /usr/bin/python3 /usr/local/bin/pypilot
             ├─1618 /usr/bin/python3 /usr/local/bin/pypilot
             ├─1626 /usr/bin/python3 /usr/local/bin/pypilot
             ├─1627 /usr/bin/python3 /usr/local/bin/pypilot
             ├─1630 /usr/bin/python3 /usr/local/bin/pypilot
             ├─1631 /usr/bin/python3 /usr/local/bin/pypilot
             └─1632 /usr/bin/python3 /usr/local/bin/pypilot

Feb 11 22:25:46 lysmarine pypilot[1618]: Settings file not found. Using defaults and creating settings file
Feb 11 22:25:46 lysmarine pypilot[1618]: Detected MPU9250/MPU9255 at standard address
Feb 11 22:25:46 lysmarine pypilot[1618]: Using fusion algorithm Kalman STATE4
Feb 11 22:25:46 lysmarine pypilot[1618]: min/max compass calibration not in use
Feb 11 22:25:46 lysmarine pypilot[1618]: Using ellipsoid compass calibration
Feb 11 22:25:46 lysmarine pypilot[1618]: Using accel calibration
Feb 11 22:25:46 lysmarine sudo[1640]:  pypilot : PWD=/home/pypilot ; USER=root ; COMMAND=/usr/bin/chrt -pi 0 1617
Feb 11 22:25:46 lysmarine sudo[1640]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Feb 11 22:25:46 lysmarine sudo[1640]: pam_unix(sudo:session): session closed for user root
Feb 11 22:25:46 lysmarine pypilot[1618]: MPU-925x init complete

````

mgrouch, 2024-02-12:
@seandepagnier 

No errors whatsoever. Another thing in OpenCPN plugin doesn't connect to pypilot unless I visit config dialog without changing anything in it

mgrouch, 2024-02-12:
In compass calibration screen of pypilot_calibration there is white stick but no blue sphere and no red cone.
When I move IMU the new dots do show up.

mgrouch, 2024-02-12:
In OpenCPN pypilot calibration it's different 

There is sphere in accelerometer screen and dots show. In compass there is small blue dot in middle and red dots show.
No sphere or cone in it. In both cases calibration doesn't start. No anything in left bottom text area

mgrouch, 2024-02-12:
In calibration plot of pypilot_web there are blue spheres for both compass and accelerometer.
Green and red dots show up. I can rotate the spheres but there is no any indication about calibration age or that
calibration is happening.

seandepagnier, 2024-02-12:
probably the calibration process is not working.

Try running pypilot from the command line.   I need to work on the logging, so that python is invoked with '-u' when running pypilot so that the log messages are not buffered for a very long time before visible.

mgrouch, 2024-02-12:
pypilot is running constantly as a daemon process. Wouldn’t those messages be flushed anyhow. It seems there were no messages reported

mgrouch, 2024-02-12:
Well I've added -u to pypilot. An there are new things:

````
user@lysmarine:~ $ sudo systemctl status pypilot@pypilot
● pypilot@pypilot.service - pypilot
     Loaded: loaded (/etc/systemd/system/pypilot@.service; enabled; preset: enabled)
     Active: active (running) since Mon 2024-02-12 04:20:26 UTC; 2min 1s ago
   Main PID: 57060 (pypilot)
      Tasks: 14 (limit: 9257)
        CPU: 3.277s
     CGroup: /system.slice/system-pypilot.slice/pypilot@pypilot.service
             ├─57060 /usr/bin/python3 -u /usr/local/bin/pypilot
             ├─57065 /usr/bin/python3 -u /usr/local/bin/pypilot
             ├─57068 /usr/bin/python3 -u /usr/local/bin/pypilot
             ├─57074 /usr/bin/python3 -u /usr/local/bin/pypilot
             ├─57076 /usr/bin/python3 -u /usr/local/bin/pypilot
             ├─57077 /usr/bin/python3 -u /usr/local/bin/pypilot
             ├─57079 /usr/bin/python3 -u /usr/local/bin/pypilot
             └─57080 /usr/bin/python3 -u /usr/local/bin/pypilot

Feb 12 04:20:32 lysmarine pypilot[57076]: signalk process 57076
Feb 12 04:20:32 lysmarine pypilot[57076]: signalk failed to read token /home/pypilot/.pypilot/signalk-token
Feb 12 04:20:32 lysmarine pypilot[57076]: signalk server found 127.0.0.1:3000
Feb 12 04:20:32 lysmarine pypilot[57076]: signalk server found 127.0.0.1:3000
Feb 12 04:20:32 lysmarine pypilot[57076]: signalk server found 127.0.0.1:3000
Feb 12 04:20:33 lysmarine pypilot[57076]: signalk found ws://127.0.0.1:3000/signalk/v1/stream?subscribe=none
Feb 12 04:20:50 lysmarine pypilot[57080]: server add socket ('127.0.0.1', 53420)
Feb 12 04:21:00 lysmarine pypilot[57065]: failed import calibration fit No module named 'scipy'
Feb 12 04:21:30 lysmarine pypilot[57065]: failed import calibration fit No module named 'scipy'
Feb 12 04:22:00 lysmarine pypilot[57065]: failed import calibration fit No module named 'scipy'

````

mgrouch, 2024-02-12:
and 

````
pip3 install scipy 
````

fixed it

Thanks

mgrouch, 2024-02-12:
python dependencies management is a mess.
How come on a system which is supposed to have more software this package was removed?

mgrouch, 2024-02-12:
I'm patching autopilot_calibration.py with this

````
20a21
> #from OpenGL.GLUT import **
52c53
<         self.Bind(wx.EVT_TIMER, lambda e : self.sHeadingOffset.SetValue(round3(self.pypilot_heading_offset)), id = self.ID_HEADING_OFFSET)
---
>         self.Bind(wx.EVT_TIMER, lambda e : self.sHeadingOffset.SetValue(round(self.pypilot_heading_offset)), id = self.ID_HEADING_OFFSET)
124c125
<                     s.SetDigits(-math.log(s.GetIncrement()) / math.log(10) + 1)
---
>                     s.SetDigits(round(-math.log(s.GetIncrement()) / math.log(10) + 1))
177c178
<         self.UpdateControl(dspin, lambda : dspin.SetValue(value))
---
>         self.UpdateControl(dspin, lambda : dspin.SetValue(round(value)))

````

It looks like there are more issues with calibrating rudder

More places to patch.

mgrouch, 2024-02-12:
Fixed in https://github.com/bareboat-necessities/lysmarine_gen/releases/tag/v2024-02-12

closed 2024-02-12 by mgrouch