bareboat-necessities/lysmarine_gen #381: influxDB error 404 with fresh 	lysmarine-bbn-bullseye_2023-08-17 install

gildas-declercq, 2024-02-15:
Hi,

I had see issue. 
as it was with an install on wich I did some updates , I did a fresh install. 
signalk is ok,
I just have a usb mouse GPS, data are ok
and a can hat , and I have the data ok from nmea2K.
I did 
sudo systemctl unmask influxdb.service
sudo systemctl enable influxdb
sudo systemctl start influxdb

as per documentation.
but cant get anything on http://localhost:8086/  except a 404 error.

what did I do wrong ?
thanks a lot
Gildas

mgrouch, 2024-02-15:
Yes, that’s right, on port 8086 you will see a 404 page. The browser will complain: “This site could not be reached” if the database is not found.

mgrouch, 2024-02-15:
and that port is without UI. it's API port.
UI for influxDb is Chronograf.

https://docs.influxdata.com/chronograf/v1/

mgrouch, 2024-02-15:
````
sudo systemctl enable chronograf
sudo systemctl start chronograf
````

And open browser http://localhost:8888/

gildas-declercq, 2024-02-15:
ok got it
thanks a lot
by the way there is a mistype in https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_influxdb
"kapasitor" instead of "kapacitor"

regards

closed 2024-03-02 by mgrouch