bareboat-necessities/lysmarine_gen #384: Never gets past initial boot sreen

Stonepaiste, 2024-02-18:
I have a problem after trying to install the new bookworm build on rasberry pi 4 using the pi touch display
I never get past the "wait for desktop to start" screen. I have been trying a lot I : 

Tried leaving it over night
booting with keyboard, mouse, ethernet 
booting without keybord, mouse, ethernet. 
Doing clean installs 4 times
Setting Wifi country codes both before first boot and after. 

After breaking off the initial boot process and restarting i'm getting a Light Display Manager Error 
It worked with the previous build. 
Got no clue on why this isn't working. 
Can i put it in some kind of debug mode. ?

mgrouch, 2024-02-18:
Looks like you applied customizations when you burned the image. It must be burned as-is without any customizations. 

https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_prepare_sd_card

Stonepaiste, 2024-02-18:
I didn’t. Burned it as it is everytime.

Everything o Changed i did after the image has been burned at did it in the config files. Also tried not changeing anything at all..
Sendt fra min iPhone

https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#_prepare_sd_card

mgrouch, 2024-02-18:
You should be able to ssh from another computer into it and check the logs for errors 

`
ssh user@lysmarine
`

Stonepaiste, 2024-02-18:
There is no logs at var/.Should they be somewhere else? Accessories the SD from another linux machine.
And how can i ssh into it if it never gets past the initial Boot screen. The one with the lysmarine text logo-(not the loading lysmarine) -and “wait for desktop to start”

mgrouch, 2024-02-18:
if you see ‘wait for desktop to boot’
OS already booted. You should be able to ssh. 

Logs are shown by

`
journalctl
`

Stonepaiste, 2024-02-18:
Couldn’t see anything other than some 3v and 12 volts warnings and that they would run on dummyregulator.

However i just downgraded to the billedet build and that works. So might be something with the build.

Also tried stripping the raspberry 4 to bare minimum without screen and with HDMI. Works on Bulseye not on Bookworm.

Sendt fra min iPhone

Stonepaiste, 2024-02-18:
I had progress.. I got it to work without the screen and connected only to HMDI after initial successful boot I added my display and then got the failed to start lighted.service again. It seems to be connected to that problem. Did you guys not run into that at all ?

Tried removing the lightdm and installing it again, but still doesn’t work, I get a message that it can’t create som config files.

mgrouch, 2024-02-19:
If lightdm service doesn’t start it wouldn’t boot into GUI. 

What are the errors you see preventing lightdm from starting? It would help to see those error messages. 

The latest images work fine here. Both lite and full are booting into GUI on pi4 and pi5. 

Thanks

Stonepaiste, 2024-02-19:
I didn’t really see anything with the lightdm in the logs, but it’s propably because i don’t know what to look for. Can i print the log or take the file from SD somewhere and send ?

I did get the error message to go away by doing this.

  *   changed the, /etc/systemd/system/display-manager.service,   to point to the lightdm service file instead of  /lib/systemd/system/;

  sudo rm /etc/systemd/system/display-manager.service

sudo ln -s /lib/systemd/system/lightdm.service /etc/systemd/system/display-manager.service

This made the error at boot go away, but still no GUI. I might be totally wrong. But it was worth a try.

could my touchdisplay be the problem?

Sendt fra min iPhone

mgrouch, 2024-02-19:
If you could burn image to SD card again as-is and

after booting

when it fails to start GUI 

do

`
systemctl  status  lightdm

`

and show output then it would help to troubleshoot the issue 

thanks

mgrouch, 2024-02-22:
Did it finally work for you?

Stonepaiste, 2024-02-22:
I guess you didn’t get my message. I sent two screenshots.
I did not get it to work unfortunatly.

I downgraded to the bullseye version of bareboat That works now.

Sendt fra min iPhone

mgrouch, 2024-02-22:
I do not see any screenshots here or in email

Stonepaiste, 2024-02-22:
Weird. Okay here is a retransfer link to them

<https://we.tl/t-yEtyJ9vpw9>
[85fc3d.jpeg]
IMG_2908.jpeg<https://we.tl/t-yEtyJ9vpw9>
we.tl<https://we.tl/t-yEtyJ9vpw9>

mgrouch, 2024-02-22:
Ok. I see lightdm error. 

What was the version of BBN OS image?

thanks

Stonepaiste, 2024-02-22:
It’s version 904 i think. i’m not next to my mac now but it was 900 something.

Best regards
Mathias

mgrouch, 2024-03-03:
Try new release

https://github.com/bareboat-necessities/lysmarine_gen/releases/tag/v2024-03-03

If it still doesn't work there, then this issue should be re-open

closed 2024-03-03 by mgrouch