bareboat-necessities/lysmarine_gen #399: OpenCPN - O-charts plugin error.

Code-Out, 2024-03-26:
When I try to activate the o-charts plugin in OpenCPN I get error:

"
The following plugin has encountered errors during startup:

o-charts

This plugin will be uninstalled. You might want to reinstall it after updating the catalogue.
"

I've tried multiple times and it's not working.

I'm running 'lysmarine-bbn-lite-bookworm_2024-03-03-r0139-raspios-arm64'

The Open-CPN version is 5.8.4.

I have this version installed on my PC and o-charts works fine.

mgrouch, 2024-03-26:
what is the output of this command:

````
md5sum /home/user/.local/lib/opencpn/libo-charts_pi.so
````

thanks

mgrouch, 2024-03-26:
On my system (BBN 2024-03-14) the output is like this: (and activates fine)

````
user@lysmarine:~ $ md5sum /home/user/.local/lib/opencpn/libo-charts_pi.so
5de885508da8818b3773331272c5467b  /home/user/.local/lib/opencpn/libo-charts_pi.so

````

If md5sum on your system is not matching you probably loaded some other version of o-charts plugin which wasn't shipped with BBN OS.

mgrouch, 2024-03-26:
For an experiment I tried to update this plugin from OpenCPN catalog.

After update the md5sum changed:

```
user@lysmarine:~ $ md5sum /home/user/.local/lib/opencpn/libo-charts_pi.so
e2e56ff4563aa7840e392e4cc85a992b  /home/user/.local/lib/opencpn/libo-charts_pi.so
````

and I got exactly same error you describe

[image]

So it looks to me it's broken in OpenCPN catalog. Original o-charts plugin included in BBN OS worked for me.

Code-Out, 2024-03-26:
Yes, it appears it was due to updating the plugin in catalog. I reinstalled the plugin from the original BBN files and it's working.

Apologies.

closed 2024-03-26 by Code-Out