bareboat-necessities/lysmarine_gen #400: OpenCPN over RealVNC slow

akerblom, 2024-04-01:
RVNC is working but OpenCPN is super slow on e.g computer while the rest of BBN OS seems to be running smooth. I’m not sure but I think it worked better initially. Any ideas on that?

mgrouch, 2024-04-01:
Is there are way for me too see how slow? 
may be post a short video?

Did you do any changes to the configs after installing?

Screenshot of output 

 top

It’s working fast and normally for me. Over VNC too.

mgrouch, 2024-04-09:
Closing as no response from user

akerblom, 2024-04-09:
Sorry have been away from my work desk, hence no response. Please re-open if possible and I’ll respond within a day.

Regards,
L

akerblom, 2024-04-09:
Hi again and thanks for the patience!

Im not sure about what configs I’ve made but I think it was initially faster and a little different in resolution. I could just reinstall the OS but I’d rather get to the “bottom” with it.

I recorded the screen with my phone while on RNVC. Maybe that’s of any use!

https://github.com/bareboat-necessities/lysmarine_gen/assets/38797235/71bc4115-8d2e-4e10-89dc-5b3cdb4b3f77

Regards,
L

mgrouch, 2024-04-09:
Is it fast when you connect a monitor without VNC?

akerblom, 2024-04-09:
Yes it works fine!

mgrouch, 2024-04-09:
What is that green triangle cursor on a video. 
I do not have that and it works fine for me. 
must have been result of config changes by you.

akerblom, 2024-04-10:
I’ll check it out. If still no luck, I’ll reinstall.

Thanks!
L

closed 2024-06-13 by mgrouch