bareboat-necessities/lysmarine_gen #407: using apt vs wget+dpkg

free-x, 2024-04-19:
Hi @mgrouch

I wonder why you use the pair wget+dpkg instead of apt/apt-get, although packages are available in repos
- avnav-ocharts*
- avnav-sailinstrument
- ais-catcher* 

Greetings
free-x

mgrouch, 2024-04-19:
I had to really optimize the build script to keep it under 1hr in CircleCi. BTW your AIS-Catcher binary is not linked to some SDRs models which are available. There were more SDRs supported than in the binaries I was packaging before that. 
Is there a reason you excluded some?

thanks

free-x, 2024-04-19:
don't know what you mean. All headers/libraries from README are there
```
$ apt install librtlsdr-dev libairspy-dev libairspyhf-dev libhackrf-dev  libsoapysdr-dev libzmq3-dev libssl-dev zlib1g-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librtlsdr-dev is already the newest version (0.6.0-4).
libairspy-dev is already the newest version (1.0.10-2+b1).
libairspyhf-dev is already the newest version (1.6.8-3).
libhackrf-dev is already the newest version (2022.09.1-3).
libsoapysdr-dev is already the newest version (0.8.1-3).
libzmq3-dev is already the newest version (4.3.4-6).
libssl-dev is already the newest version (3.0.11-1~deb12u2).
zlib1g-dev is already the newest version (1:1.2.13.dfsg-1).
```

mgrouch, 2024-04-19:
I was comparing outputs of

````
AIS-Catcher -L
````

of that 
with this version I built awhile back

https://github.com/bareboat-necessities/lysmarine_gen/releases/download/vTest/AIS-catcher-20231216-bookworm-arm64.zip

free-x, 2024-04-20:
seems to be cmake bug

free-x, 2024-04-20:
Спасибо, починил ;-)
```
$ obj-aarch64-linux-gnu/AIS-catcher -L
AIS-catcher (build Apr 17 2024) v0.56-122-gf99160f8
(C) Copyright 2021-2023 jvde-github and other contributors
This is free software; see the source for copying conditions.There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
SDR support: RTLSDR AIRSPY AIRSPYHF+ RTLTCP SPYSERVER ZMQ HACKRF 
Other support: SSL ZLIB RTLSDR-BIASTEE RTLSDR-TUNERBW 
```

mgrouch, 2024-04-22:
> Спасибо, починил ;-)
>
> ```
> $ obj-aarch64-linux-gnu/AIS-catcher -L
> AIS-catcher (build Apr 17 2024) v0.56-122-gf99160f8
> (C) Copyright 2021-2023 jvde-github and other contributors
> This is free software; [...]

Your fix to AIS-catcher binary made it into 2024-04-20 version of BBN Marine OS.

Thanks

closed 2024-04-22 by mgrouch

free-x, 2024-04-22:
you have closed issue, and what has changed?
https://github.com/bareboat-necessities/lysmarine_gen/commit/985edd917dc58d1d6e23de6e5b168bbae0cd14f0

wget+dpkg is still used instead of apt

mgrouch, 2024-04-22:
That’s not an issue