bareboat-necessities/lysmarine_gen #414: Python version outdated - HomeAssistant

metzger100, 2024-05-05:
I get a Warning in HomeAssistant:

> Support for Python 3.11.2 will be removed
This will no longer work in version 2024.4. Please fix this before upgrading.
Support for running Home Assistant in the currently used Python version 3.11.2 is deprecated and will be removed in Home Assistant 2024.4. Please upgrade Python to 3.12 to prevent your Home Assistant instance from being corrupted.

I also can't install HACS. After executing the install script as the homeassistant user:

```bash
sudo su
su - homeassistant
wget -O - https://get.hacs.xyz | bash -
```

After adding HACS as integration the error is like that:

> Logger: homeassistant.config_entries
Quelle: config_entries.py:2444
Erstmals aufgetreten: 22:39:12 (1 Vorkommnisse)
Zuletzt protokolliert: 22:39:12

> Error occurred loading flow for integration hacs: cannot import name 'keys' from 'securesystemslib' (/srv/homeassistant/lib/python3.11/site-packages/securesystemslib/__init__.py)

It seems like it has to do with the outdated python version:

https://www.reddit.com/r/homeassistant/comments/1cimhub/tried_to_enable_hacs_on_fresh_installation_got/

Not sure how I can fix that. On Arch linux python packages are managed by pacman but it looks like pip manages python on BBN OS. Would be glad if you could help me with that issue.

PS: I really love that OS it is very comfortable and easy to use.

mgrouch, 2024-05-05:
Have you tried the suggestion from that reddit thread

`
pip3 install securesystemslib==0.31.0
1.0 does not work
`

as homeassistant user

metzger100, 2024-05-06:
Yes I tried that. But that didn't help.

mgrouch, 2024-05-06:
In Reddit post they have this issue on python 3.12 too. So the issue exists even with recommended python version. 
I’d suggest checking on HomeAssistant specific support forums.

metzger100, 2024-05-06:
I didn't find much. I have to open a new thread. That might take some time. I will update this Issue.

metzger100, 2024-05-20:
Update: There is a Issue on the Home Assistant site now:
https://github.com/home-assistant/core/issues/117451

Then I noticed that my Home Assistant Instance is outdated and lagging several versions behind.
BBN OS: 2024.3.3
recent Version: 2024.5.4

How do I update HomeAssistant on BBN OS?

metzger100, 2024-05-20:
Update 2: I fixed it.

How did I fix it?

I logged in as root-user changed the password of the homeassistant user, because I didn't have a password for it and made it a sudo user:

```bash
sudo su
```

```bash
passwd homeassistant
```

```bash
usermod -aG sudo homeassistant
```

I left the rootuser-CLI by pressing CTRL-D. After that I followed the instruction of [this guide](https://community.home-assistant.io/t/python-3-12-backport-for-debian-12-bookworm/709459) to install a backport of python3.12:

Switch to the user that is running Home Assistant and install python3.12:

```bash
sudo -u homeassistant -H -s
```

```bash
wget -qO- https://pascalroeleven.nl/deb-pascalroeleven.gpg | sudo tee /etc/apt/keyrings/deb-pascalroeleven.gpg
```

```bash
cat <<EOF | sudo tee /etc/apt/sources.list.d/pascalroeleven.sources
Types: deb
URIs: http://deb.pascalroeleven.nl/python3.12
Suites: bookworm-backports
Components: main
Signed-By: /etc/apt/keyrings/deb-pascalroeleven.gpg
EOF
```

```bash
sudo apt update
```

```bash
sudo apt install python3.12 python3.12-venv python3.12-dev
```

```bash
sudo apt-get install -y python3 python3-dev python3-venv python3-pip bluez libffi-dev libssl-dev libjpeg-dev zlib1g-dev autoconf build-essential libopenjp2-7 libtiff6 libturbojpeg0-dev tzdata ffmpeg liblapack3 liblapack-dev libatlas-base-dev
```

```bash
cd /srv/homeassistant
```

```bash
python3.12 -m venv .
```

```bash
source bin/activate
```

```bash
python3.12 -m pip install wheel
```

```bash
pip3.12 install homeassistant==2024.5.4
```

Then I logged out of homeassistant user (CTRL-D) and rebooted:

```bash
reboot
```

After that it took a while till Home Assistant started with the version 2024.5.4. Then the installation of HACS worked flawlessly.

metzger100, 2024-05-20:
I also install https://home-assistant-china.github.io/docs/ecosystem/hass-configurator/
To have an webbased editor for the homeassistant configuration files.

```bash
cd /home/homeassistant/.homeassistant/
```

```bash
wget https://raw.githubusercontent.com/danielperna84/hass-configurator/master/configurator.py
```

```bash
sudo chmod 755 configurator.py
```

Adjust the BASEPATH to `BASEPATH = "/home/homeassistant/.homeassistant/"`:

```bash
nano configurator.py
```

Then enable the autostart of the webeditor via systemd:

```bash
sudo nano /etc/systemd/system/hass-configurator.service
```
Paste that into that file:

```bash
[Unit]
Description=Home Assistant Configurator
After=network.target

[Service]
Type=simple
User=homeassistant
ExecStart=/usr/bin/python3.12 /home/homeassistant/.homeassistant/configurator.py
WorkingDirectory=/home/homeassistant/.homeassistant
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

activate the service:

```bash
sudo systemctl enable --now hass-configurator.py
```

The editor is now available on http://BBN-OS-IP-Address:3218/

That is needed because the homeassistant mqtt signalk integration is outdated and broken.
I followed this guide to setup mqtt entities via the configuration yaml
https://community.home-assistant.io/t/bridging-signalk-server-nmea-and-home-assistant-using-signalk-mqtt-bridge-and-mosquitto-mqtt/642640/1

mgrouch, 2024-05-20:
Thanks. It’s helpful. 
Could you please share more HomeAssistant sensors configuration yaml files so others could just copy and paste?
There is a need for some universal SignalK sensors definition file. 

Thanks again!

metzger100, 2024-05-20:
I can't. There is no way you can copy and paste it because it depends on the SignalK Configuration.

For example the freezer temprature:

```yaml
mqtt: # SignalK MQTT HA Bridge
  # https://developers.home-assistant.io/docs/core/entity/sensor/
  sensor:
    - name: "SignalK MCS Kuehlschrank Temperatur"
      unique_id: "signalk_mcs_freezer_temperature"
      state_topic: N/signalk/bb8e1eedbfb1/vessels/self/environment/inside/freezer/temperature
      value_template: "{{ value_json.value - 273.15 }}"
      unit_of_measurement: "C"
```

The state_topic depends on your config in SignalK. I bought a MCS Board to get some Fieldbuses to the Raspberry. 

[image]

I configured the 1-Wire-Sensor as a freezer temperature in SignalK with the topic path "vessels/self/environment/inside/freezer/temperature"
There hundreds of possible configurations. Even if we could gather all configurations for Home Assistant together, there would be hundreds of "dead" Elements in Home Assistant. Then there would still be the problem of the ID in MQTT. SignalK generates a random ID for MQTT. In my case its bb8e1eedbfb1. It gets displayed in the SignalK-mqtt-bridge Plugin. You always need to adjust that. And if you are a weird US american how wants his/her freezer temperature in Fahrenheit you lost because the value_template is completely different. I can write a guide about setting up a mqtt element of the SignalK mqtt-broker like I did with the configurator and the python backport. But there will always be some understanding of mqtt needed to configure the sensors.

mgrouch, 2024-05-20:
Ok. After pasting they could do global replace of it in a text editor. What would help is all paths specified by SignalK exposed as HomeAssistant sensors in one file. 

Thanks

metzger100, 2024-05-20:
Yeah. The structure SignalK sends to the mqtt-broker is always kind of the same:

[image]

closed 2024-06-13 by mgrouch