bareboat-necessities/lysmarine_gen #416: Lightdm service fails to start after installing DSI connected screen drivers

mgrouch, 2024-05-07:
Lightdm service fails to start after installing DSI connected screen drivers 

users reported it as an issue with edatec
ED-HMI3020-101C

Waveshare DSI LCD

See also 
https://github.com/bareboat-necessities/lysmarine_gen/discussions/327#discussioncomment-6546423

mgrouch, 2024-05-07:
[image]

mgrouch, 2024-05-07:
Also

https://www.cruisersforum.com/forums/f134/recommended-cockpit-platform-285146-10.html#post3896976

mgrouch, 2024-05-09:
Per Edatec support after the following changes lightdm starts:

removed /etc/udev/rules.d/99-uinput.rules and modified /usr/share/X11/xorg.conf.d/90-touchinput.conf

````
user@lysmarine:~ $ cat /usr/share/X11/xorg.conf.d/90-touchinput.conf
Section "InputClass"
    Identifier "calibration"
    Driver "evdev"
    MatchProduct "4-0014 Goodix Capacitive TouchScreen gt911 Synopsys DesignWare I2C adapter"
    MatchDevicePath "/dev/input/event*"
    Option "EmulateThirdButton" "1"
    Option "EmulateThirdButtonTimeout" "750"
    Option "EmulateThirdButtonMoveThreshold" "30"
EndSection
````

mgrouch, 2024-05-09:
Regarding Waveshare DSI touchscreen issues

here is a thread on raspberry pi forums:

https://forums.raspberrypi.com/viewtopic.php?t=366011

mgrouch, 2024-05-09:
Rotating DSI display discussion
https://forums.raspberrypi.com/viewtopic.php?t=369626

mgrouch, 2024-05-16:
Regarding screen rotation on Edatec

https://www.cruisersforum.com/forums/f134/cant-install-opencpn-on-rpi5-286085.html#post3899802

mgrouch, 2024-06-04:
Edatec support was able to solve all issues with their touchscreen on BBN OS:

Hi Mikhail,

I've solved the problem of twofing not working.

It needs to modify “/usr/share/X11/xorg.conf.d/90-touchinput.conf”

 
````
Section "InputClass"
    Identifier "calibration"
    Driver "evdev"
    MatchProduct "Goodix Capacitive TouchScreen"
    MatchDevicePath "/dev/input/event*"
    Option "EmulateThirdButton" "1"
    Option "EmulateThirdButtonTimeout" "750"
    Option "EmulateThirdButtonMoveThreshold" "30"
    Option "Calibration" "0 1280 0 800"
    Option "SwapAxes" "1"
    Option "Inversion" "1 0"
EndSection

````

mgrouch, 2024-11-22:
More on DSI displays 

https://github.com/raspberrypi/linux/issues/6341

Turduli, 2025-01-26:
Hi everyone,

I believe I’m experiencing the same issue, but unfortunately, the suggested fix hasn’t worked for me.

Here’s my setup and what I’ve tried so far:

I’ve updated everything, including the EEPROM, to the most recent versions.
I’m using the official Raspberry Pi Touch Display 2 with BBN OS.
When I connect the display, the system fails to boot and shows the error:
"Failed to start Lightdm.service".
I also noticed that the 90-touchinput.conf file is missing from the /usr/share/X11/xorg.conf.d/ directory, which prevents me from modifying it as suggested in other solutions.
Has anyone successfully gotten the Raspberry Pi Touch Display 2 to work with BBN OS? Any guidance or ideas would be greatly appreciated!

Thank you!

mgrouch, 2025-01-27:
> Hi everyone,
> 
> I believe I’m experiencing the same issue, but unfortunately, the suggested fix hasn’t worked for me.
> 
> Here’s my setup and what I’ve tried so far:
> 
> I’ve updated everything, including the EEPROM, to the most recent versions. I’m using the official Raspberry Pi Touch Display 2 with BBN OS on a RP 5. When I connect the display, the system fails to boot and shows the error: "Failed to start Lightdm.service". I also noticed that the 90-touchinput.conf file is missing from the /usr/share/X11/xorg.conf.d/ directory, which prevents me from modifying it as suggested in other solutions. Has anyone successfully gotten the Raspberry Pi Touch Display 2 to work with BBN OS? Any guidance or ideas would be greatly appreciated!
> 
> Thank you!

If you have DSI connected screen you are dependent of your screen vendor. You ask the vendor for the driver for exact version of the Linux kernel you are running. Or (better) using HDMI connected screen.