bareboat-necessities/lysmarine_gen #493: SignalK install skips percompiling some modules

mgrouch, 2025-02-24:
Errors look like:

````
Failed to start: Could not locate the bindings file. Tried: ...
````

mgrouch, 2025-02-24:
Running 

````
pnpm build
````

as 'signalk' user in the directory specified in the error message usually fixes the issue.

mgrouch, 2025-02-25:
This issue is caused by introduction of --approve-builds flag to pnpm.
Even worse  --approve-builds pops up a interractive menu in CI which hangs the builds.

https://github.com/pnpm/pnpm/issues/9102

mgrouch, 2025-02-25:
fixed in 2025-02-25

closed 2025-02-25 by mgrouch

maelp, 2025-02-25:
How did you fix it?

maelp, 2025-02-25:
apparently a better fix is to use the "neverBuiltDependencies" field in
package.json

Message ID: ***@***.***
.com>

mgrouch, 2025-02-25:
I can't edit package.json. I receive the code which I'm just building without modifying it