bareboat-necessities/lysmarine_gen #515: Ideas for easy upgrade to new version

jsextl, 2025-04-18:
Hi BBN team,

I use BBN since about 1 year now and really like your approach and the feature set.

However I always struggle to do the upgrades - as far as I understand, upgrade is only supported by downloading a new image and then start from scratch with the configuration - yes I know, I can copy config files, etc. and have no problem with that being a linux expert.

My problem is, that I do not know, which files I have to change as things in the GUI or SignalK web UI will change files in the background and for me there is no easy way to find out which files are changed and where the configuration is stored.

I also tried the upgrade / hotfix procedure outlined in the documentation:

cd /home/user/add-ons
./hot-fixes-install.sh

But that does not seem to do anything on my system.

apt-get update / dist-upgrade seems to upgrade the OS packages, but not components like SignalK.

So I feel pretty lost even being a linux/server expert and I think that the approach you are taking is just a very special one as normally you can upgrade Debian based systems without problems from one version to the other even for major upgrades - yes, sometimes it is necessary to fix the configuration for some packages but this is normally clearly stated during the upgrade process.

So maybe I'm missing something or there are ideas how other people are handling their upgrades? Or just leave the system running as is?

Thanks a lot for your input!

Jonas

mgrouch, 2025-04-19:
SignalK is not packaged as Debian package so you can’t upgrade it like usual Debian components. PyPilot is another one which doesn’t follow Debian packaging.

jsextl, 2025-04-21:
Ok I see - used OpenPlotter in the past and maybe they packaged it on their own? Because there it was always possible to upgrade all packages without reinstallation.

However - does anybody have good ideas how to automate the configuration? Or sample-scripts? Maybe using Ansible? I could script those things easily but I have no idea how to get the information about which files I have to modify if I add things in the SignalK WEB GUI for example - where are they stored? In json files? In a database? Central configuration file? That's my missing link :-)

How are other users doing this? Just never upgrade? Or repeat all configuration in the web interface each time?

mgrouch, 2025-04-21:
The way updates are done in OpenPlotter is the common source of things getting broken. 
You could see it in OP forum which is full of reports of something no longer working after some update. 

SignalK configuration is stored in /home/signalk/.signalk/

usually in JSON format

jsextl, 2025-04-21:
Well it was not so bad - I was using it for long time and had no problems with minor upgrades - from 3 to 4 it was a bigger step.

All right - will have a look into this and if I get some scripts ready I can share with others.